realsenseai/librealsense #10999: If my kernel is 5.17.13 how can i install SDK

Glean-jcy, 2022-10-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-10-15:
Hi @Glean-jcy  If you are unable to change your kernel version to one that is officially supported by the librealsense SDK then there is the option of building the SDK from source code with CMake with the **RSUSB backend** installation method.  This type of build bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require patching.  It is best suited to single-camera projects rather than multiple cameras on the same computer though.

Glean-jcy, 2022-10-16:
> Hi @Glean-jcy If you are unable to change your kernel version to one that is officially supported by the librealsense SDK then there is the option of building the SDK from source code with CMake [...]
THX！

MartyG-RealSense, 2022-10-25:
Hi @Glean-jcy  Do you require further assistance with this case, please?  Thanks!

Glean-jcy, 2022-10-27:
> 

oh i can run realsense-viewer now,but i dont now what i had done,and i am doing other things now ,so thx your reply!!!

MartyG-RealSense, 2022-10-27:
It's excellent to hear that realsense-viewer is now working for you.  Thanks very much for the update!

Glean-jcy, 2022-10-27:
> It's excellent to hear that realsense-viewer is now working for you. Thanks very much for the update!

I feel surprised that my kernel has always been 5.17.13, and I've never changed it.

MartyG-RealSense, 2022-10-27:
The SDK can work with unsupported kernels, though there may be unpredictable consequences in regards to stability.  It seems as though it is stable on your particular computer though.

Glean-jcy, 2022-10-27:
> The SDK can work with unsupported kernels, though there may be unpredictable consequences in regards to stability. It seems as though it is stable on your particular computer though.

Now It seems stable.

MartyG-RealSense, 2022-10-27:
Some kernel versions of the same general kernel type but different version numbers may be more stable than others.  A past example was that some RealSense users found kernel **5.11.0-34-generic** to be stable but slightly newer kernels such as 5.11.0-**38**-generic had problems.

Glean-jcy, 2022-10-27:
> Some kernel versions of the same general kernel type but different version numbers may be more stable than others. A past example was that some RealSense users found kernel **5.11.0-34-generic** to be stable but slightly newer kernels such as 5.11.0-**38**-generic had problems.

I learned a lot, thanks!

closed 2022-11-09 by MartyG-RealSense