realsenseai/librealsense #11006: Uninitialized memory in rs2_project_color_pixel_to_depth_pixel

stefanhige, 2022-10-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    | --                                      | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0 | 
| Language                            |  python                          | 
| Segment			| -- | 

### Issue Description

Python utility function rs2_project_color_pixel_to_depth_pixel defined here

https://github.com/IntelRealSense/librealsense/blob/af7510c8d58f70381056a758cd3f4553aa37fc69/wrappers/python/pyrsutil.cpp#L52

returns possibly uninitialized memory from line
https://github.com/IntelRealSense/librealsense/blob/af7510c8d58f70381056a758cd3f4553aa37fc69/wrappers/python/pyrsutil.cpp#L45
if the equivalent cpp function rs2_project_color_pixel_to_depth_pixel iterates through a trajectory that maps to all-zero depth values.
https://github.com/IntelRealSense/librealsense/blob/af7510c8d58f70381056a758cd3f4553aa37fc69/src/rs.cpp#L3772

This also happens in practice in my application.

Possible fix would be
`std::array<float, 2> to_pixel{-1, -1};`
or similar, indicating to the caller that no valid pixel could be found.

MartyG-RealSense, 2022-10-18:
Hi @stefanhige  If you would like to suggest a code change to the RealSense SDK's implementation of the **rs2_project_depth_pixel_to_color_pixel** then you can submit a Pull Request (PR) that makes the suggested change to **rs.cpp**.

https://github.com/IntelRealSense/librealsense/pulls

If you require the change immediately then you could also create your own custom version of librealsense (either by downloading the source code, or by 'forking' librealsense onto your own GitHub account) and then edit rs.cpp and build the customized SDK.

Nir-Az, 2022-10-24:
Hi @stefanhige ,
Thanks for your contribution, your PR was merged 🥇 

This issue can be closed.

MartyG-RealSense, 2022-10-29:
Hi @stefanhige  May this issue be closed now that your PR has been merged?  Thanks!

stefanhige, 2022-11-01:
Sure! Thank you.

closed 2022-11-01 by stefanhige