realsenseai/librealsense #11007: Intel RealSense Viewer Access Violation!

GaoChangTai, 2022-10-19:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2022-10-19:
Hi @GaoChangTai   Do you experience this error if you download and run the standalone version of the RealSense Viewer - **Intel.RealSense.Viewer.exe** (version 2.51.1) - from the 'Assets' file list at the link below, please.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

Once downloaded, Intel.RealSense.Viewer.exe should be launched with a double-leftclick from the folder location on your computer that it was downloaded to and not the Viewer launch shortcut icon on the Windows desktop.

GaoChangTai, 2022-10-19:
> 

> Hi @GaoChangTai Do you experience this error if you download and run the standalone version of the RealSense Viewer - **Intel.RealSense.Viewer.exe** (version 2.51.1) - from the 'Assets' file list at the link below, please.
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1
> 
> Once downloaded, Intel.RealSense.Viewer.exe should be launched with a double-leftclick from the folder location on your computer that it was downloaded to and not the Viewer launch shortcut icon on the Windows desktop.
> 
> Please also check the camera privacy settings in Windows to ensure that permission to access the camera hardware and permission for applications to access the camera are both set to **On**. Instructions for doing so can be found at [#3381 (comment)](https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789)

Your method worked very well. Thank you very much！！！

GaoChangTai, 2022-10-19:
> 

Your method worked very well. Thank you very much.

GaoChangTai, 2022-10-19:
> Hi @GaoChangTai Do you experience this error if you download and run the standalone version of the RealSense Viewer - **Intel.RealSense.Viewer.exe** (version 2.51.1) - from the 'Assets' file list at [...]

Your method worked very well. Thank you very much.

GaoChangTai, 2022-10-19:
> Hi @GaoChangTai Do you experience this error if you download and run the standalone version of the RealSense Viewer - **Intel.RealSense.Viewer.exe** (version 2.51.1) - from the 'Assets' file list at [...]

@MartyG-RealSense Your method worked very well. Thank you very much!

MartyG-RealSense, 2022-10-19:
That's excellent news, @GaoChangTai - thanks very much for the update!  :)

MartyG-RealSense, 2022-10-25:
Case closed due to solution achieved and no further comment received.

closed 2022-10-25 by MartyG-RealSense