realsenseai/librealsense #11009: Depth Frames not sending Frames

pkyleUPI, 2022-10-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Ubuntu 20.04 | 
| Kernel Version (Linux Only)   | 5.4.0-188 | 
| Platform                                 | PC |
| SDK Version                          |  v2.51.1           | 
| Language                              |  C++                | 
| Segment			        |     Robot          | 

### Issue Description
I've browsing the issues for some time now and havn't been able to find a solution that works for me.. Any help would be greatly appreciated.

### When I turn on the Stereo Module:
- the Depth Stream fails to start and I get a "No Frames Recieved!" error in the player. 

### Output from 'modinfo uvcvideo | grep "version:"'
version:        1.1.2.realsense-1.3.18
srcversion:     17500165C02F9242C508B66

### DMESG OUTPUT
#### When I Plug In camera before starting realsense-viewer:
usb 4-2: new SuperSpeed Gen 1 USB device number 4 using xhci_hcd
usb 4-2: New USB device found, idVendor=8086, idProduct=0b5c, bcdDevice=50.d0
usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 4-2: Product: Intel(R) RealSense(TM) Depth Camera 455 
usb 4-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 455 
usb 4-2: SerialNumber: 110223060813
uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:20/0000:20:08.1/0000:22:00.3/usb4/4-2/4-2:1.0/input/input782
uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
hid-sensor-hub 0003:8086:0B5C.0011: No report with id 0xffffffff found

#### When I start realsense-viewer
uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:20/0000:20:08.1/0000:22:00.3/usb4/4-2/4-2:1.0/input/input783
uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:20/0000:20:08.1/0000:22:00.3/usb4/4-2/4-2:1.0/input/input784
uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:20/0000:20:08.1/0000:22:00.3/usb4/4-2/4-2:1.0/input/input785
uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:20/0000:20:08.1/0000:22:00.3/usb4/4-2/4-2:1.0/input/input786
uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:20/0000:20:08.1/0000:22:00.3/usb4/4-2/4-2:1.0/input/input787
.... This continues...

#### When I turn on Stereo Module 
No new messages to Dmesg feed... 

### RealSense-Viewer Debug Console-Window Output:
Starting streaming of Depth...
sensor.cpp:1480 - Request: Z16 Depth, 
Resolved to: Z16 Depth, 
uvc-streamer.cpp:28 - endpoint 130 read buffer size: 815104
messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
types.h:898 - Right MIPI error
notifications.cpp:511 - Right MIPI error
messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
types.h:898 - Depth stream start failure
notifications.cpp:511 - Depth stream start failure

...This continues and also will display similar errors for Left MIPI

MartyG-RealSense, 2022-10-20:
Hi @pkyleUPI  I note that your D455 is being detected by DMESG and the serial number is correctly read.  

Would it be possible for you to try launching the **rs-capture** RealSense example program to confirm whether the problem is with the RealSense Viewer tool specifically or RealSense programs in general, please?

pkyleUPI, 2022-10-20:
rs-capture did not appear to work, here is the output to the terminal:
20/10 08:20:24,295 ERROR [140521112590080] (ds5-thermal-monitor.cpp:94) Error during thermal compensation handling: query is available during streaming only
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x556754fbc840):
    Frame didn't arrive within 15000

MartyG-RealSense, 2022-10-20:
Thank you very much for the information.  The messages **No Frames Received** and **Frame didn't arrive within 15000** indicate that the camera is not receiving frames and is 'timing out' after a certain amount of seconds has passed after the start of streaming without frames being received. 

Could you provide information about the following questions, please.

1.  Are you using the official 1 meter long USB cable supplied with the camera or a longer USB cable of your own choice?

2.  Which method did you use to install the librealsense SDK (packages, source code, etc)?

pkyleUPI, 2022-10-20:
Here's the output when I re-installed the pre-built package this morning (ubuntu 20.04, focal. Kernel: 5.4.0-128):

sudo apt-get install librealsense2-dkms
[sudo] password for pkyle: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  microcode-initrd
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  librealsense2-udev-rules
The following NEW packages will be installed:
  librealsense2-dkms librealsense2-udev-rules
0 upgraded, 2 newly installed, 0 to remove and 1 not upgraded.
Need to get 0 B/4,283 kB of archives.
After this operation, 23.0 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 436766 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.18-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.18-0ubuntu1) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.51.1-0~realsense0.7526_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.51.1-0~realsense0.7526) ...
Setting up librealsense2-udev-rules:amd64 (2.51.1-0~realsense0.7526) ...

Postinst script activated...
Presets deployment:
no src dir
tgt dir exists
Presets deployment is skipped
Setting up librealsense2-dkms (1.3.18-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.18 DKMS files...
Building for 5.4.0-128-generic 5.15.0-52-generic
Building initial module for 5.4.0-128-generic
Secure Boot not enabled on this system.
Done.

uvcvideo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-128-generic/updates/dkms/

videodev.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-128-generic/updates/dkms/

hid-sensor-accel-3d.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-128-generic/updates/dkms/

hid-sensor-gyro-3d.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-128-generic/updates/dkms/

depmod...

DKMS: install completed.
Building initial module for 5.15.0-52-generic
Error!  The /var/lib/dkms/librealsense2-dkms/1.3.18/5.15.0-52-generic/x86_64/dkms.conf for module librealsense2-dkms includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.

Loading the modified modules into kernel...modprobe: FATAL: Module hid_sensor_accel_3d is in use.
Current status:
librealsense2-dkms, 1.3.18, 5.4.0-128-generic, x86_64: installed
nvidia, 510.85.02, 4.15.0-188-generic, x86_64: installed
nvidia, 510.85.02, 5.11.0-46-generic, x86_64: installed
nvidia, 510.85.02, 5.15.0-52-generic, x86_64: installed
nvidia, 510.85.02, 5.4.0-128-generic, x86_64: installed
nvidia, 510.85.02, 5.4.0-131-generic, x86_64: installed
nvidia, 510.85.02, 5.8.0-63-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.3.18
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms

pkyleUPI, 2022-10-20:
On this most recent install, the RGB Camera works but i receive the following (new) Dmesg output:

[Thu Oct 20 11:47:03 2022] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

Depth still does not work. - right MIPI error and depth stream start failure, then "No Frames recieved"

modinfo uvcvideo | grep "version:"
version:        1.1.2.realsense-1.3.18
srcversion:     17500165C02F9242C508B66

pkyleUPI, 2022-10-20:
In further research it might be prudent for me to point out I'm using an AMD Ryzen Threadripper 3960x

MartyG-RealSense, 2022-10-21:
RealSense 400 Series cameras can work with any Intel or Arm processor.  There have been a few past cases in which Ryzen Threadripper has been reported to work with 400 Series cameras whilst other AMD CPU brands did not.

Quality is important with own-choice USB cables used with RealSense cameras due to the high volume of data bandwidth that they can transmit through the cable.  The history of support cases on this forum also demonstrates that there is an increased chance of degraded performance or no camera connection when using USB C to C cables rather than the official cable's A to C.

As far as I am aware, the continuously changing input number in dmesg has not previously been reported by other RealSense users.

Your **sudo apt-get install librealsense2-dkms** installation log looks mostly okay to me.  I note the error **FATAL: Module hid_sensor_accel_3d is in use** near the end.  A RealSense team member provides advice at https://github.com/IntelRealSense/librealsense/issues/7436 for resolving it by restarting Ubuntu in a case where kernel 5.4 was also being used.

pkyleUPI, 2022-10-21:
Marty
Our team plugged in a D435 that we had and this resolved all of our issues. at this point we believe it is an issue with our specifc D455 camera. I will close this case.

closed 2022-10-21 by pkyleUPI