realsenseai/librealsense #11010: Announcement: RealSense D457 GMSL/FAKRA stereo depth camera now in stock at the RealSense Store

MartyG-RealSense, 2022-10-20:
Hi everyone,

Intel's new **RealSense D457** stereo depth camera model is now listed as in stock at the official online RealSense Store.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d457.html

A range of information resources about this camera can be found at https://github.com/IntelRealSense/librealsense/issues/10964

Please note that the D457 model is designed for use with a **GMSL/FAKRA** connection and cannot be used as a "plug and play" USB device.

sam598, 2022-10-29:
@MartyG-RealSense Is this camera capable of synchronized color and depth exposures? Or is it the same as all of the other RealSense cameras where "RGB and depth frames have a temporal offset bounded by a period of one frame"?

If it is the latter, it is very surprising that Intel would sell another camera without fixing this critical issue for 3D scanning. It cannot be understated how much this issue has prevented the success of RealSense hardware over the years.

MartyG-RealSense, 2022-10-29:
@sam598  I will discuss your D457 sync question with my Intel RealSense colleagues.  Thank you.

MartyG-RealSense, 2022-10-29:
Hi again @sam598   My colleagues informed me that the depth and RGB are synced in D457 as they are on the same PDB, unlike D435.

sam598, 2022-10-29:
Hi @MartyG-RealSense , to be absolutely clear I am not talking about Genlock. I am talking about color and depth synchronization on a single camera.

With your comment here: https://github.com/IntelRealSense/librealsense/issues/10850#issuecomment-1232999740 and this discussion here: https://github.com/IntelRealSense/librealsense/issues/10926#issuecomment-1295874596 it is clear that the other D400 cameras are not capable of Color and Depth sync.

"RGB and depth frames have a temporal offset bounded by a period of one frame" is not exposure synchronization. Any camera pair is capable of temporal alignment. For 3D scanning it is a requirement that the color and depth images be exposed at the exact same moment in time.

Is this a feature that has been added to the D457 that was not present in the D455?

MartyG-RealSense, 2022-10-29:
The genlock reference in the provided response is because I highlighted your interest in genlock sync to my colleagues whilst discussing your D457 query.

I will seek clarification about whether the D457 PDB advice relates to depth and RGB on the same camera.

MartyG-RealSense, 2022-10-30:
Hi @sam598  My colleagues confirmed that it is referring to sync between the depth and RGB sensors on the same D457 camera. 
 RGB and depth are on the same PCB.

sam598, 2022-10-30:
Thanks. Before I purchase dozens more of these cameras I need to ask that you can verify these two statements are true:

1) The D415, D435, D435i, and D455 cameras are not capable of color and depth sync because the cameras are printed on separate PCB. Instead the SDK matches frames based on their hardware time stamps, and there is a temporal offset of one frame or less.

2) The D457 camera has its color and depth cameras printed on the same PCB, and is able to synchronously capture color and depth frames at the exact same moment in time. There is no offset in the hardware time stamps, because they are being internally triggered at the same time.

sam598, 2022-10-30:
I don’t understand how the D455 and D457 cameras can both use the same D450 camera module and not both suffer the same issue.

MartyG-RealSense, 2022-10-30:
I have referred your latest questions to my colleagues.  Thanks again for your patience.

sam598, 2022-10-30:
@MartyG-RealSense I may have found an answer.

In the [D457 datasheet](https://www.intelrealsense.com/wp-content/uploads/2022/10/D457-Datasheet-092022.pdf) on section 7.13:
`Intel® RealSense™ D400 Series supports hardware sync signal for multi-camera configuration. Multiple cameras are hardware synchronized as to capture at identical times and frame rates. For multiple D457s, all of them can work in slave mode;  the external trigger is applied to the camera through the GMSL cable from the host side. There are two triggers, one for depth camera and one for RGB camera. The depth and RGB cameras can be triggered at both the same and different supported frame rates.`

In the [D400 family datasheet](https://www.intelrealsense.com/wp-content/uploads/2022/03/Intel-RealSense-D400-Series-Datasheet-March-2022.pdf) on section 7.13:
`Intel® RealSense™ D400 Series supports a hardware sync signal for multi-camera configuration. For multiple cameras to be hardware synchronized so as to capture data at identical times and frame rates, pins 5 (SYNC) and pins 9 (Ground) on the external sensor sync connector will need to be connected. The external sensor sync connector is located on the Vision Processor D4 V1 and V3 board and is accessible on the Depth Cameras.  D405 does not support the hardware sync signal for multi-camera configuration. `

It does indeed seem like this is a new feature that has been added to the D457. I have additional question, mainly which pin on the D457 connector can be used to trigger the color camera. I can open up a new issue if that makes sense instead?

MartyG-RealSense, 2022-10-30:
Thanks very much @sam598 for the update.  Let's stay on this issue.  I have added your third question to my communication to my colleagues.

sam598, 2022-11-02:
Any update on which pin can be used to trigger RGB?

MartyG-RealSense, 2022-11-03:
I received a response from my colleagues about your question about which pin on the D457 connector can be used to trigger the color camera.  The separate triggers referred to in your D457 data sheet quote are from the GMSL cable, not the 9-pin connector.  This may therefore make the feature irrelevant for your needs if you plan to make use of the 9-pin connector.

khuongkhoa1999, 2022-11-04:
Hi,
Our product in development has been currently using D415/D435 for a while now and we would like to have a camera with an IP rating and secure connectors since ours is an outdoor agricultural application.
Everything has been written with Realsense SDK. GMSL/FAKRA is a new concept for us but we hope to integrate the camera quickly. 
Can we keep using RealSense SDK with the GMSL interface? Is there any guide for the GMSL interface? We are currently using Jetson NX as our host platform but that may change in the future.

MartyG-RealSense, 2022-11-04:
Hi @khuongkhoa1999  Yes, you can continue to use the SDK.  There is a pre-release test version of the RealSense SDK (2.52.1) with GMSL support.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.52.1

More information about cabling is can be found at https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1290843233 and the comments beneath it.

In addition to the D457 camera, you will need a **Maxim Deserializer** component on the host side to decode the streams.  You could initially test the D457 quickly without needing a deserializer though by using it with Connect Tech's products, and then later switch to using the deserializer component when you need to integrate the D457 into your outdoor weatherproofed system.

khuongkhoa1999, 2022-11-07:
Thanks @MartyG-RealSense 

I see that you have added MIPI support, can I just buy an off-the-shelf deserialize board like LI-GMSL2-IPX-DESER, which has MAX9296A Deserializer, uses the MIPI output and have the same plug-and-play capabilities like with USB?
https://www.leopardimaging.com/product/accessories/adapters-carrier-boards/li-gmsl2-ipx-deser/

One more question, we were led to believe Intel is slowly shutting down production and support for its 3D camera business but since the release of this camera, we want to give this one more chance before exploring different alternatives. What are the near-future plans for the D400 series in 10-year time?

MartyG-RealSense, 2022-11-07:
I will consult with my Intel RealSense colleagues regarding your deserializer question.

The reports about the future of RealSense were a miscommunication by news websites.  Whilst some RealSense camera models were declared End Of Life (retired), the 400 Series stereo camera models have continued.  Intel's official statement is below.

https://github.com/IntelRealSense/librealsense#update-on-recent-changes-to-the-realsense-product-line

MartyG-RealSense, 2022-11-18:
Hi @khuongkhoa1999  Do you require further assistance, please?  Thanks!

MartyG-RealSense, 2022-11-24:
Does anyone have further questions on this discussion about D457, please?  Thanks!

haitaoQ, 2022-11-29:
Hi @MartyG-RealSense,

Is there a detailed guide for using D457 on AGX Xavier?
I have D457 & 9296 De-serializer Board & Nvidia AGX Xavier. Install SDK based on pre-release version v2.52.1. But rs-enumerate-devices doesn't detect any cameras.

Looking forward to your reply. Thanks!

MartyG-RealSense, 2022-11-29:
Hi @haitaoQ  I have referred your question to my Intel RealSense colleagues. Thanks very much for your patience!

RealSenseCustomerSupport, 2022-11-29:
----------------------------------------------

Hi @haitaoQ, there is a reference driver package designed for D457 to work with Intel De-serializer board on the AGX Xavier. Please kindly submit a request through Zendesk support portal for us to provide the driver under NDA. Thank you.

closed 2022-12-12 by MartyG-RealSense

AndreV84, 2023-01-03:
@RealSenseCustomerSupport How to contact you through zendesk via NDA? Could you extend, please?

AndreV84, 2023-01-07:
@haitaoQ @khuongkhoa1999 
Were you able to get the d457 with LI-GMSL2-IPX-DESER working on Jetson AGX as plug and play?
did it require to "port the driver" ?

jan-krueger, 2023-02-14:
> In addition to the D457 camera, you will need a **Maxim Deserializer** component on the host side to decode the streams. You could initially test the D457 quickly without needing a deserializer though by using it with Connect Tech's products, and then later switch to using the deserializer component when you need to finally integrate the D457 into your outdoor weatherproofed system.

Good morning, I would like to confirm a couple of things. I have been reading this thread but I am still a bit confused.

Assuming that the AGX Xavier carrier/board I choose has the Maxim Deserializer included is this basically just plug-and-play/does Intel offer the necessary Linux drivers to make the D457 work through librealsense? Why is the deserializer necessary when it is possible to test the camera without it, or is the testing just possible through a different port that is not GMSL/FAKRA?

If the driver for the D457 on the AGX Xavier exists where would I find it? 

Thank you for your time.

MartyG-RealSense, 2023-02-14:
Hi @jan-krueger  The D457 is not plug and play like the other RealSense 400 Series models are.  It requires a 'reference driver'.  Intel provide a driver for use with AGX Xavier at the link below.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

The D457 has a USB port in addition to the GMSL / FAKRA interface, though it is intended for use for debugging purposes or production line testing rather than normal everyday use.  The D457 is designed to be used with a GMSL / FAKRA connection.  

The de-serializer component decodes the streams at host-side.  If the D457-compatible product by **Connect Tech** is used then it is not necessary to purchase a sepaate de-serializer as it is built into Connect Tech's hardware. 

https://connecttech.com/connect-tech-offers-built-in-support-on-the-newly-announced-intel-gmsl-realsense-depth-camera-d457/

AndreV84, 2023-02-14:
I am not certain why would not it work as plug and play given the OS of the nvidia Jetson has the driver for the de-serializer
https://docs.nvidia.com/jetson/l4t-multimedia/group__max9295.html
from that point of view it is not clear what exactly needs to be done with the reference driver in order to get the camera working e.g. with intel de-serializer, in my opinion

MartyG-RealSense, 2023-02-14:
There are instructions at the link below for installing the D457 driver to AGX Xavier.

https://github.com/IntelRealSense/realsense_mipi_platform_driver#install-kernel-and-d457-driver-to-jetson-agx-xavier

The driver was designed for Intel's own de-seriaizer product though.

MartyG-RealSense, 2023-02-14:
Regarding using the camera with USB, new advice has just been posted by an Intel Customer Support team member.

https://support.intelrealsense.com/hc/en-us/community/posts/10293079498387/comments/13946561028243

jan-krueger, 2023-02-14:
@MartyG-RealSense Thank you for your swift response. In one of your previous comments you said "In addition to the D457 camera, you will need a Maxim Deserializer component on the host side to decode the streams.". As far as I can tell Maxim belongs to Analog Devies and is not an Intel subsidiary/related. If I understand it correctly that comment implies that the Intel driver works with the Maxim Deserializer *and* the Deserializer that Intel offers?

MartyG-RealSense, 2023-02-14:
It works with the Intel de-serializer, and products that contain Maxim de-serializer technology.  An example of such products that has been discussed is those by a company called **Leopard Imaging**.  

https://www.leopardimaging.com/product/accessories/adapters-carrier-boards/li-gmsl2-ipx-deser/

My Intel RealSense colleagues confirmed that the product in the above link can be used with D457.

My understanding is that de-serializer products containing **MAX9296A** technology are compatible, as discussed at https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1363709490

MartyG-RealSense, 2023-02-16:
Hi everyone, I have received information from my Intel RealSense colleagues that it is fine to use the D457 only in USB mode and it has the same USB performance and the same feature-set as the D455, with the added IP65 protection.  So it is absolutely valid to purchase the D457 for the protection and use it as a USB camera instead of in GMSL / FAKRA mode.

masip85, 2024-08-12:
Hello

https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues/221#issuecomment-2259853655

Is possible to have this project working for orin nano dev kit? I've tried to use it directly, or adapting it modifying files , but I really don't know how to do this. I'd really need some help and guidance. Why is orin nano/nx outside the scope of this project? Does it need many  changes?

MartyG-RealSense, 2024-08-12:
Hi @masip85  The MIPI driver is designed for AGX Xavier **and** Intel's own brand of deserializer board.

https://store.intelrealsense.com/buy-intel-realsense-des457.html

When using D457 with a different model of Jetson or a different deserializer board, the customer needs to 'port' the driver for use with the particular equipment that they have.  Intel are unable to assist customers with this porting process, unfortunately.  

It looks as though the MIPI driver's installation instructions for Orin are expecting JetPack 6 to be installed.  Is your Orin Nano using JP6, please?

https://github.com/IntelRealSense/realsense_mipi_platform_driver?tab=readme-ov-file#install-kernel-and-d457-driver-to-jetson-orin

Nvidia advise in a discussion at the link below that if JetPack 5.1 is used on Orin then a manual fix has to be applied in order for the “Configure Jetson AGX CSI Connector” option to work.  They included the fix in JP versions after 5.1.

https://forums.developer.nvidia.com/t/r35-2-1-jetson-io-configure-jetson-agx-csi-connector/241893