realsenseai/librealsense #11011: How can I use the realsense tool in matlab to record the video and record it to the target folder

Confident2000, 2022-10-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Operating System & Version |   {Win (7)   |  
| Platform                                 | PC  |
| SDK Version                            |  SDK 2.0                        | 
| Language                            |  matlab       | 
### Issue Description
How can I use the realsense tool in matlab to record the video and record it to the target folder, so that I can later stream it as output to get the captured depth frame data?
There may be a case of c, but I would like to ask if there is a matlab version

MartyG-RealSense, 2022-10-20:
Hi @Confident2000  the link below provides scripting for recording camera data to a .bag format file at a specified file path using the RealSense MATLAB wrapper.

https://support.intelrealsense.com/hc/en-us/community/posts/6846011994003-How-do-I-record-a-bag-file-using-the-MATLAB-RealSense-wrapper-

Confident2000, 2022-10-20:
Hi@MartyG-RealSense Thank you very much for your help. I will try according to the reference you provided

Confident2000, 2022-10-20:
I used the following code according to the reference you provided, but there was an error. I'm sorry to bother you again. Please help me have a look

```
% Make Config object to manage pipeline settings
cfg = realsense.config();
% Tell pipeline to record to the given rosbag file
cfg.enable_record_to_file('E:\Realsense_project\RS_bag');
% Make Pipeline object to manage streaming
pipe = realsense.pipeline();
pointcloud = realsense.pointcloud();
% Start streaming from the rosbag with default settings
profile = pipe.start(cfg);
% Get streaming device's name
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);
% Main loop
for i = 1:10
% Obtain frames from a streaming device
fs = pipe.wait_for_frames();
% Select depth frame
depth = fs.get_depth_frame();
%color = fs.get_color_frame();
end
% Stop streaming
pipe.stop();
```
----------------------------------
错误使用 librealsense_mex
Error opening file: E:\Realsense_project\RS_bag

出错 realsense.pipeline/start (第 31 行)
                    out = realsense.librealsense_mex('rs2::pipeline', 'start', this.objectHandle, varargin{1}.objectHandle);

出错 realsense.getbag (第 13 行)
profile = pipe.start(cfg);

MartyG-RealSense, 2022-10-20:
Hi @Confident2000  I note that you are using the file path **E:\Realsense_project\RS_bag**

The bag filename should end in .bag

So if your file is called RS_bag then change the path to E:\Realsense_project\RS_bag.bag

If the file is just called RS then change the path to E:\Realsense_project\RS.bag

MartyG-RealSense, 2022-10-25:
The MATLAB documentation link below provides information about registering two point clouds and stitching them using **Iterative Closest Point (ICP)**.

https://uk.mathworks.com/help/vision/ug/3-d-point-cloud-registration-and-stitching.html

closed 2022-11-01 by Confident2000