realsenseai/librealsense #11016: L515 Ｎot Ｄetected

FeiSong123, 2022-10-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |      L515     | 
| Firmware Version                   |      | 
| Operating System & Version |   Ubuntu18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-131-generic             | 
| Platform                                 | PC |
| SDK Version                            |             2.51.1                         | 
| Language                            |  C++                      | 
| Segment			|  Robot                | 

### Issue Description
<Describe your issue / question / feature request / etc..>

The viewer cannot detect the camera. USB3.0 port didn't detect the camera. A computer plugged into the USB3.0 port cannot be detected by the viewer, but the RGB camera can be used as an ordinary camera for other software. When plugged into the USB2.0 port, the viewer can detect, but the display is 2.1 instead of 3.1. Strangely, when the deep stream, color stream and motion are turned on, they can be displayed normally.The USB cable is 3.0. It has been tested and has no problems.

MartyG-RealSense, 2022-10-22:
Hi @FeiSong123  Could you first try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

FeiSong123, 2022-10-22:
I tried it because you mentioned this phenomenon to me last time. But I tried several times and it didn't work.

MartyG-RealSense, 2022-10-22:
Thank you.  When a RealSense camera is plugged into a USB 2 port or uses a USB 2 cable, it will be detected as a USB 2 device.  If an own choice of USB cable is used, a USB 2 cable may sometimes be accidentally used instead of a USB 3 cable.

FeiSong123, 2022-10-22:
But my current situation is that the port where the camera is plugged into USB 3.0 doesn't work. I can detect it with another camera, but I have a camera that can't detect it.

MartyG-RealSense, 2022-10-22:
Is the camera that is detected in the USB 3 port an ordinary RGB webcam and not a RealSense camera?

FeiSong123, 2022-10-22:
Yes, but viewer can't detect it. The camera software that comes with Windows can detect the RGB camera and use it.

MartyG-RealSense, 2022-10-22:
An RGB webcam would not appear automatically in the Viewer like a RealSense camera does.  Such a camera can be detected and added to the Viewer by clicking the **Add Source** button at the top of the Viewer's side-panel and then selecting **Platform Camera**.

FeiSong123, 2022-10-22:
But I want to be able to use the camera normally, depth flow, color flow and motion flow as before.

Now I can't detect anything in the viewer.

FeiSong123, 2022-10-22:
_Another problem with the USB 2.0 port is that it can turn on depth stream, color stream and motion stream normally. But I got an error when I used the program to call the camera._

MartyG-RealSense, 2022-10-22:
Only RGB can be accessed when a camera is accessed via the Platform Camera option.

What error message did you get, please?

FeiSong123, 2022-10-22:
No, I don't mean access the camera through the platform. Normally the camera would be plugged into the computer and the viewer would be able to detect those sensors, but there are none.

I am not in the lab now, so I can't read the error message. I will check it tomorrow when I go to the lab. But that error is the camera USB 2.0 will report error.

MartyG-RealSense, 2022-10-22:
Do you recall if the error might have been **Runtime error: Couldn't resolve requests**

FeiSong123, 2022-10-22:
I didn't try again, and I can't remember exactly what it was. The only thing I remember is that the error was reported when the camera was plugged into the 2.0 USB port. 

The exact error will have to wait until tomorrow.

MartyG-RealSense, 2022-10-22:
Okay, thanks very much.  I look forward to your next report.  Thanks for the information provided so far!

FeiSong123, 2022-10-23:
> 

I get the error below:

terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
Aborted (core dumped)

Can you tell me how to add pictures to my reply,  please ?

MartyG-RealSense, 2022-10-23:
You can add pictures to your reply by copying an image with Ctrl-C and then pasting it into the comment box with Ctrl-V.  

**Couldn't resolve requests** can mean that a particular stream configuration (such as 848x480 RGB at 30 FPS) that was requested is not currently supported by your particular camera model.  This can occur if the camera is on a USB 2.1 connection, because fewer resolution / FPS combinations are available on USB 2.1 than when the camera is on a USB 3 connection.

**USB 3**

[image]

**USB 2.1**

[image]

FeiSong123, 2022-10-23:
> No, I don't mean access the camera through the platform. Normally the camera would be plugged into the computer and the viewer would be able to detect those sensors, but there are none.
>
> I am not in [...]

[image]
This is the viewer when camera connected with USB2.0 cable, and it's detected.

But when I try with USB3.0 cable, viewer display nothing.

FeiSong123, 2022-10-23:
Could it be that the camera is broken ? Because my other camera is normal.

MartyG-RealSense, 2022-10-23:
So on a USB 2.1 connection in the Viewer, an L515 will be limited to 320x240 depth and 640x360 or 640x480 RGB.  

Only being able to obtain USB 2.1 mode on an L515 can be an indication of a hardware fault in that particular camera unit.  If it is a hardware fault then the only action available other than to only use 2.1 mode is to return the camera to the retailer that you purchased it from if it is less than a year old.

FeiSong123, 2022-10-23:
Well, it seems that my camera has a hardware failure. I'll contact my retailer.

Another problem is that I have a microcomputer. Every time I turn on the computer, I can't detect the camera. I need to plug and unplug the USB port for a minute or two to detect the camera.

MartyG-RealSense, 2022-10-23:
For some Linux users, **rtcwake** (putting the computer to sleep for a certain period of seconds and then waking it up) helped with the problem of the camera not being detected during boot, as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/6123#issuecomment-603171387

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/5913#issuecomment-590936871 stated that using `rtcwake -m off -s 30` resolved the problem in their particular case.

More information about using rtcwake can be found at the link below.

https://www.addictivetips.com/ubuntu-linux-tips/automatically-wake-linux-up-from-sleep/

FeiSong123, 2022-10-24:
Sometimes when I restart the computer to open the program, he will report the following error:

terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

I have read https://github.com/IntelRealSense/librealsense/issues/2850, but I'm not sure how to use it. My kernel here is 5.4.0-131-generic.

FeiSong123, 2022-10-24:
Sometimes when I restart the computer to open the program, he will report the following error, just plug and unplug the USB would be fine. But it's a bit troublesome to do this every time you start up.

terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

I have read https://github.com/IntelRealSense/librealsense/issues/2850, but I'm not sure how to use it. My kernel here is 5.4.0-131-generic.

MartyG-RealSense, 2022-10-29:
Hi @FeiSong123  Do you have any progress to report about your issue, please?  Thanks!

FeiSong123, 2022-10-29:
No, you can close it. Thanks for your help !

closed 2022-10-29 by MartyG-RealSense