realsenseai/librealsense #11020: Whether there are any papers about the error analysis about D455 and L515?

jiejie567, 2022-10-25:
I'll be appreciated if you offer me some information.

MartyG-RealSense, 2022-10-25:
Hi @jiejie567  Could you tell me more about what information you need about error analysis, please?  Thanks!

RealSenseCustomerSupport, 2022-10-25:
----------------------------------------------

Hello Zendesk,

Engineering has identified the hardware issue with the type-C USB connector when inserting the cable in one of the two possible orientations, resulting in enumeration in USB 2.0 mode. When inserting the type-C connector, the USB 2.0 pins make contact first and if inserted relatively slowly (<100 ms) the detection circuit that determines the routing of the USB 3.0 signals has a lag time before the signals connect to the DS5 if it has to flip the multiplexer setting; during that lag time the DS5 sees the USB 2.0 connection starts operation in USB 2.0.

This issue is being addressed in the next board revision with the redesign having been successfully tested on a development platform (improved multiplexer).

In the interim, as long as the plug is inserted quickly or if the host side (type A) is plugged in last, it should always enumerate in USB 3.0.

Also ensure that the type-C cable is not a charging only cable as these ONLY have the USB 2.0 connections.

Regards,
MartyG.

jiejie567, 2022-10-25:
@MartyG-RealSense hi MartyG, nice to get your reply again. I want to the the mathematical model about error of the range measurement. For example, I think the measuremen terror of D455 will increase with the distance increase, then what is the specific law of change?

MartyG-RealSense, 2022-10-25:
Hi @jiejie567  You are correct that error increases linearly over distance.  This is called **RMS error**.  Information about this can be found in point 5 of the section of Intel's camera tuning guide linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#verify-performance-regularly-on-a-flat-wall-or-target

[image]

Whilst there is not a graph describing RMS error for D455 like the D415 versus D435 one below (though one can be plotted with the formula in that linked-to section), the curve for D455 RMS should be similar to the green D415 lower line.  This is because D415 depth error over distance is around 2x less than D435, and the D455 has 2x the accuracy over distance of the D435.

[image]

You can obtain a real-time RMS value for both D455 and L515 using the RealSense SDK's **Depth Quality Tool**.

[image]

The depth accuracy over distance for the L515 is quoted in its RealSense Store listing as **~5 mm to ~14 mm through 9 meters**

MartyG-RealSense, 2022-10-31:
Hi @jiejie567  Do you require further assistance with this case, please?  Thanks!

jiejie567, 2022-10-31:
Thanks, MartyG. I will close this issue.

closed 2022-10-31 by jiejie567