realsenseai/librealsense #11024: Meaning of x and y in rs2_deproject_pixel_to_point

DaveBGld, 2022-10-25:
I need to obtain the position in 3D of a pixel in a D435i camera frame to create a new pose in ROS for a robot arm end effector.

After obtaining the color and depth frames, and locating the pixel of interest, I am coding: 

```
rs2_intrinsics intrinsics;
intrinsics = selection.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>().get_intrinsics();
float pixel[2];
float my3Dpoint[3]; // x, y, z
pixel[0] = sideA.x;  // column
pixel[1] = sideA.y;  // row
float depthofPoint = depth.get_distance (sideA.x, sideA.y);
rs2_deproject_pixel_to_point (my3Dpoint, &intrinsics, pixel, depthofPoint);
```

In ROS, x is forward, y is left. What is **x** and **y** in my3Dpoint? I am assuming z is depth What is the meaning of negative values on these? 

Also, do I need to use here the depth stream instrinsics?

Many thanks!!

David

MartyG-RealSense, 2022-10-26:
Hi @DaveBGld  RealSense cameras use a coordinate system where Y is vertical and Z (depth) is forward, as you correctly assume, whilst in the ROS convention Z is vertical.  More information about this can be found at https://github.com/IntelRealSense/realsense-ros/issues/1099

RealSense coordinate systems specify coordinates in terms of whether they are to the left, right, above or below the center-line of a sensor.  So an X or Y value may be positive or negative depending on where the coordinate is relative to the sensor center-line.  https://github.com/IntelRealSense/librealsense/issues/7279#issuecomment-689031488 discusses this.

Depth intrinsics are typically used in RealSense ROS scripts for obtaining the Z-depth value.  At https://github.com/IntelRealSense/realsense-ros/issues/807#issuecomment-510359646 advice is provided for obtaining Z by using **rs2_deproject_pixel_to_point**.

DaveBGld, 2022-10-26:
Thanks @MartyG-RealSense ..

I actually found [here](https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#point-coordinates) that

_"... Each stream of images provided by this SDK is also associated with a separate 3D coordinate space, specified in meters, with the coordinate [0,0,0] referring to the center of the physical imager. Within this space, the positive x-axis points to the right, the positive y-axis points down, and the positive z-axis points forward. Coordinates within this space are referred to as "points", and are used to describe locations within 3D space that might be visible within a particular image... "_

As I progress I have a different problem with imprecise values from rs2_deproject_pixel_to_point, but I'll post that on a separate issue to keep the history clean...

Thanks!

closed 2022-10-26 by DaveBGld