realsenseai/librealsense #11029: Align the processed color_frame with the depth_frame

yanyanxinshi, 2022-10-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win (10)   | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC |
| SDK Version                            | 2.51.1  | 
| Language                            |  python                        | 
| Segment			|  others                      | 

### Issue Description
hello, I want to align the processed color_frame with the depth_frame. After searching a lot of information, I can't find how to do it. Can you help me? thank you very much!

MartyG-RealSense, 2022-10-26:
Hi @yanyanxinshi  The RealSense SDK has a Python depth to color alignment example program called **align_depth2color.py** at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

As you mention the **processed** color_frame though, are you applying some kind of adjustment to the color data, please?

yanyanxinshi, 2022-10-26:
> Hi @yanyanxinshi The RealSense SDK has a Python depth to color alignment example program called **align_depth2color.py** at the link below.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py
> 
> If you wanted to perform color to depth alignment instead of depth to color then change 'align_to = rs.stream.color' to **align_to = rs.stream.depth**
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L60
> 
> As you mention the **processed** color_frame though, are you applying some kind of adjustment to the color data, please?

Yes, you are right. The color data I am referring to is the color data after white balance. Can I still align the color data with the depth data after this processing?

MartyG-RealSense, 2022-10-26:
It should not matter whether color white-balance is adjusted before or after alignment, as it is not changing the position or scale of the pixels, just their color intensity.

If **post-processing** filters such as Decimation were being applied to the color stream then alignment should be applied after post-processing.

yanyanxinshi, 2022-10-26:
As you can see I'm new to realsense and when I've got the image `# Get frameset of color and depth#
         frames = pipeline.wait_for_frames()` , how do I get the color frame from the image frame? Thanks!

MartyG-RealSense, 2022-10-26:
In the align_depth2color.py script, the quoted instruction is two separate lines.

# Get frameset of color and depth
frames = pipeline.wait_for_frames()

The # sign makes a line a comment, meaning that it will be ignored when the script is run.

In your quote, you do:

# Get frameset of color and depth
# frames = pipeline.wait_for_frames()

If there is a # in front of 'frames' then this line will be ignored, making th script work incorrectly.  So if you have **# frames = pipeline.wait_for_frames()** in your script then the # on that line should be removed.

MartyG-RealSense, 2022-10-26:
You asked "How do I get the color frame from the image frame?".  Could you provide more information about what you are aiming to do, please?

yanyanxinshi, 2022-10-26:
> 

Yes, I found the error I quoted, I mean, after I use `frames = pipeline.wait_for_frames()`， the next step is "Align the depth frame to color frame"，`aligned_frames = align.process(frames)`， how can I make it in "Align the depth frame" to color frame" before entering the color frame into my white balance code? Thanks a lot for your answer!

yanyanxinshi, 2022-10-26:
> 你问 “如何从图像框架中获取颜色框架”。你能提供更多关于你打算做什么的信息吗？

Of course, I want to apply the camera to a low light scene where the captured RGB image is not detailed enough, so I plan to enhance it with a white balance algorithm, but then combine the enhanced RGB frame with the depth frame align

MartyG-RealSense, 2022-10-26:
I believe that the best place to insert the white-balance setting instruction would be on the line directly after your script's pipeline start instruction (not after wait_for_frames).

In a low light situation though, increasing RGB **Brightness** may make more difference than increasing white balance.

**Default brightness**

[image]

**Maximum brightness**

[image]

yanyanxinshi, 2022-10-26:
> 

Thanks for your answer, I will try it

yanyanxinshi, 2022-10-27:
Thank you for your hard work, I have solved the problem, thank you for your help

closed 2022-10-27 by yanyanxinshi

MartyG-RealSense, 2022-10-27:
That's great news that you achieved a solution, @yanyanxinshi - thanks very much for the update  :)