realsenseai/librealsense #11030: Use the V4L Native backend by applying the kernel patching,in rk3588

lantianhaohh, 2022-10-26:
Run the script (note the ending characters - L4T)

./scripts/patch-realsense-ubuntu-L4T.sh  

How to create a kernel patch in rk3588, run the above sentence, you will not find /etc/nv_tegra_release file,

MartyG-RealSense, 2022-10-26:
Hi @lantianhaohh  For an RK device I would recommend building librealsense from source code using the **RSUSB backend** installation method, which bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require patching.  RSUSB is also a useful install method for computers / computing devices with software or hardware that has a non-typical configuration, such as industrial microcomputers.

Instructions for an RSUSB build of librealsense can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

MartyG-RealSense, 2022-10-31:
Hi @lantianhaohh  Was the above comment helpful, please?  Thanks!

MartyG-RealSense, 2022-11-12:
I refer readers of this case to https://github.com/IntelRealSense/librealsense/issues/11050

closed 2022-11-12 by MartyG-RealSense