realsenseai/librealsense #11032: D405 post processing parameters

HarisSiv, 2022-10-26:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D405 } | 
| Firmware Version                   | (v2.51.1 | 
| Operating System & Version |   {Windows 11  |                                     | 
| Platform                                 | PC |
| SDK Version                            |  2.51.1 }                          | 
| Language                            |  LabView }                          | 
| Segment			|  { Others }                          | 

### Issue Description

Hello,

We are trying to save the the parameters that we've setup on our D405 camera through Intel RealSense Viewer v2.51.1.

When we save the parameters in a JSON file, we can see all the parameters associated to "Controls" and "Advanced Controls" in the file, but we get none of the parameters linked to "Depth Visualization" and "Post-Processing".

We then read the parameters setup in the JSON file in LabView. Since we are missing "Depth Visualization" and "Post-Processing" data in the JSON file, we try to adjust our values manually to get an image as close as what we had in the Intel RealSense Viewer.

Is there a reason why those parameters aren't saved in the JSON file? Is there a way to save them or to access them?

MartyG-RealSense, 2022-10-26:
Hi @HarisSiv  Not all of the RealSense Viewer tool's settings, like the ones that you quoted, can be exported to json file.  An example of this is the **Histogram Equalization** depth visualization setting, which is not a camera hardware feature and so cannot be exported to json.  Instead, Histogram Equalization is applied through the depth **colorizer** feature that color-shades depth data according to the depth visualization settings.

Post-processing filters are also software-based features that are applied to camera data in real-time and processed on the computer's CPU, not in the camera hardware.

Settings that cannot be defined in a json file can be configured with RealSense SDK program scripting in languages such as C++ and Python.

Further information about this subject can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4441403077395-Realsense-viewer-Ability-to-preset-settings-

https://github.com/cansik/realsense-processing/issues/24 is also a helpful reference.

HarisSiv, 2022-10-27:
Thank you, much appreciated

closed 2022-10-27 by HarisSiv