realsenseai/librealsense #11039: What is the smallest movement that L515  can detect

Anjing44, 2022-10-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello,from the datasheet of L515,it said that the depth accuracy is <5mm @1m，and <14mm@9m, and the ideal range of L515 is 0.25m to 9m, so if the depth is 0.25m, what is the accuracy

MartyG-RealSense, 2022-10-28:
Hi @Anjing44  There is not an official figure for accuracy at 0.25 m distance.  However, in regard to the smallest trackable movement, a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/4175#issuecomment-507448389 estimated that for a 400 Series camera, the minimum trackable size of a moving object should be 9 pixels large.  I would assume a similar value to apply to the L515.

Anjing44, 2022-10-29:
> Hi @Anjing44 There is not an official figure for accuracy at 0.25 m distance. However, in regard to the smallest trackable movement, a RealSense team member at [#4175 (comment)](https://github.com/IntelRealSense/librealsense/issues/4175#issuecomment-507448389) estimated that for a 400 Series camera, the minimum trackable size of a moving object should be 9 pixels large. I would assume a similar value to apply to the L515.
> 
> As a guideline to how large 9 pixels is, the resolution of the stream will tell you how many pixels the image has in total. For example, the L515's 1024x768 resolution would be an image 1024 pixels wide and 768 pixels high.

Thanks so much

closed 2022-11-12 by MartyG-RealSense