realsenseai/librealsense #11044: realsense D405 failed with opencv official sample of videocapture_openni.cpp

viisautta, 2022-10-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405} | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Fedora 36  | 
| Kernel Version (Linux Only)    |  6.0.5  | 
| Platform                                 | PC |
| SDK Version                            |  2.52.1 | 
| Language                            |  cpp  | 
| Segment			|  opencv  | 

### Issue Description
[o@fedora bin]$ ./example_cpp_videocapture_openni 
Device opening ...
[RS2] +Rs2Driver
[RS2] initialize 
[RS2] enumerateDevices 
[RS2] devicesChanged removed=(nil) added=0x13292d0
[RS2] deviceConnected
[RS2] queryDeviceInfo 
[RS2] DEVICE serial=218622272939 name=Intel RealSense D405
[RS2] Rs2Driver INITIALIZED
[RS2] deviceOpen uri=218622272939
[RS2] +Rs2Device
[RS2] initialize 
[RS2] queryDeviceInfo 
[RS2] DEVICE serial=218622272939 name=Intel RealSense D405
[RS2] initializeStreams 
[RS2] SENSOR 0
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=1280 height=720 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=1280 height=720 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=1280 height=720 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=5
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=10 width=1288 height=808 framerate=25
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=10 width=1288 height=808 framerate=15
[... 281 lines cut ...]
[RS2] start type=1 sensorId=0 streamId=0
[RS2] updateConfiguration 
[RS2] getSensorInfoList 
[RS2] stop type=1 sensorId=0 streamId=0
[RS2] updateConfiguration 
[RS2] destroyStream ptr=0x13395a0
[RS2] deviceClose ptr=0x1330290
[RS2] ~Rs2Device
[RS2] shutdown 
[RS2] ~Rs2Stream type=3
[RS2] ~Rs2Stream type=3
[RS2] ~Rs2Stream type=1
[RS2] shutdown type=1 sensorId=0 streamId=0
[ERROR:0@2.761] global /home/o/Public/Devel/opencv-4.6.0/modules/videoio/src/cap.cpp (278) open VIDEOIO(OPENNI2): raised OpenCV exception:

OpenCV(4.6.0) /home/o/Public/Devel/opencv-4.6.0/modules/videoio/src/cap_openni2.cpp:383: error: (-2:Unspecified error) in function 'toggleStream'
> CvCapture_OpenNI2::CvCapture_OpenNI2 : Couldn't find color stream: 	Device: Can't find this source 2
> 	Context: Couldn't create stream from device:0132c620, source: 2
> 

done.
Can not open a capture object.
[RS2] shutdown 
[RS2] ~Rs2Driver
[o@fedora bin]$ pwd

MartyG-RealSense, 2022-10-30:
Hi @viisautta  Have you set up the RealSense **OpenNI2 wrapper** using the instructions for Linux at the link below so that a RealSense camera can be used with OpenNI2 example programs, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2

**********

RealSense SDK version 2.52.1 would not be suitable for use with the D405 camera model as 2.52.1 is a 'pre-release' version designed for use with cameras with **GMSL** connector support such as the new RealSense D457.  For your D405 USB camera I recommend using SDK version 2.51.1.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

*********

The Fedora kernel 6.05 is not officially supported by the RealSense SDK.  The SDK can work with unsupported kernels but there may be unpredictable consequences in regards to stability.  I would recommend building the RealSense SDK from source code using the **RSUSB backend** installation method, which bypasses the kernel and so is not dependent on Linux versions of kernel versions and does not require patching.  

An example of instructions for using the RSUSB installation method can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

viisautta, 2022-10-30:
@MartyG-RealSense 
The RSUSB backend installation method has been used, and the OpenNI2 wrapper of SimpleViewer does not work.
I will study how to downgrade SDK version to 2.51.1.

MartyG-RealSense, 2022-10-30:
Thanks very much @viisautta for the update.  I look forward to your next report after the SDK downgrade.  Good luck!

MartyG-RealSense, 2022-10-30:
The mention of not being able to find the color stream suggests that this could be a D405-specific issue.  The D405 model does not have a separate RGB sensor, unlike other 400 Series camera models such as D415, D435 and D455, and instead obtains RGB from the depth sensor.

I will discuss this case to my Intel RealSense colleagues.  Thanks very much for your patience!

viisautta, 2022-10-30:
@MartyG-RealSense 
retry the openni2 sample, its output is as below:
[o@192 Bin]$ ./SimpleViewer 
[RS2] +Rs2Driver
[RS2] initialize 
[RS2] enumerateDevices 
[RS2] devicesChanged removed=(nil) added=0x8ad370
[RS2] deviceConnected
[RS2] queryDeviceInfo 
[RS2] DEVICE serial=218622272939 name=Intel RealSense D405
[RS2] Rs2Driver INITIALIZED
After initialization:

[RS2] deviceOpen uri=218622272939
[RS2] +Rs2Device
[RS2] initialize 
[RS2] queryDeviceInfo 
[RS2] DEVICE serial=218622272939 name=Intel RealSense D405
[RS2] initializeStreams 
[RS2] SENSOR 0
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=1280 height=720 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=1280 height=720 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=1280 height=720 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=480 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=640 height=360 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=480 height=270 framerate=5
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=90
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=60
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=30
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=15
[RS2] 	type=3 sensorId=0 streamId=0 profileId=5 format=5 width=424 height=240 framerate=5
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=10 width=1288 height=808 framerate=25
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=10 width=1288 height=808 framerate=15
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=1280 height=720 framerate=30
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=1280 height=720 framerate=15
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=1280 height=720 framerate=5
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=5
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=480 framerate=5
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=360 framerate=90
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=360 framerate=60
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=360 framerate=30
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=360 framerate=15
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=360 framerate=5
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=480 height=270 framerate=90
[RS2] 	type=3 sensorId=0 streamId=1 profileId=5 format=9 width=480 height=270 framerate=60
[... 255 lines cut ...]
[RS2] 	type=3 sensorId=0 streamId=2 profileId=6 format=9 width=424 height=240 framerate=5
[RS2] getSensorInfoList 
[RS2] createStream sensorType=3
[RS2] mainLoop 
[RS2] start type=1 sensorId=0 streamId=0
[RS2] updateConfiguration 
[RS2] getSensorInfoList 
SimpleViewer: Couldn't find color stream:
	Device: Can't find this source 2
	Context: Couldn't create stream from device:008b04b0, source: 2

SimpleViewer: No valid streams. Exiting
[RS2] stop type=1 sensorId=0 streamId=0
[RS2] updateConfiguration 
[RS2] destroyStream ptr=0x8943a0
[RS2] deviceClose ptr=0x898070
[RS2] ~Rs2Device
[RS2] shutdown 
[RS2] ~Rs2Stream type=3
[RS2] ~Rs2Stream type=3
[RS2] ~Rs2Stream type=1
[RS2] shutdown type=1 sensorId=0 streamId=0
[RS2] shutdown 
[RS2] ~Rs2Driver
[o@192 Bin]$

MartyG-RealSense, 2022-10-30:
Thanks @viisautta for the further log.  I will update you as soon as I have further information to provide.

MartyG-RealSense, 2022-11-12:
Discussions with my Intel RealSense colleagues about this case are ongoing.  Thanks very much @viisautta for your patience!

MartyG-RealSense, 2022-11-15:
Hi @viisautta  My colleagues have advised that there are no plans to add D405 support to the OpenNI2 wrapper in the near future, unfortunately.  Yourself or another RealSense OpenNI2 wrapper user are welcome though to submit a Pull Request (PR) to add that feature to the wrapper. 

https://github.com/IntelRealSense/librealsense/pulls

MartyG-RealSense, 2022-11-21:
Hi @viisautta   Bearing in mind the above comment, do you require further assistance please?  Thanks!

closed 2022-11-27 by MartyG-RealSense