realsenseai/librealsense #11046: export an xyz file

1215232494, 2022-10-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |   Linux (Ubuntu 20)   | 
| Language                            |  python                        | 

### Issue Description
How do I export an xyz file, similar to using `pyrealsense2.points.export_to_ply()` export ply file in Python?

MartyG-RealSense, 2022-10-31:
Hello, if you are aiming to export a color xyzrgb ply file then the only Python script for doing so with export_to_ply that is known to work is at https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293

If you do not require color to be included in the ply file then the RealSense SDK has an export_to_ply example program called **export_ply_example.py** at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

1215232494, 2022-10-31:
Hi @MartyG-RealSense , thanks for your reply.I got the ply file with color by using [`points.export_to_ply`](https://github.com/IntelRealSense/librealsense/blob/923253f645e867c75e3c6c30a84127f41d82f3e9/wrappers/python/examples/pyglet_pointcloud_viewer.py#L491), but it is a binary file. How do I get the value of xyzrgb from it?Or whether non binary files can be generated, such as using [`ply.set_option(rs.save_to_ply.option_ply_binary, False)` ](https://github.com/IntelRealSense/librealsense/blob/923253f645e867c75e3c6c30a84127f41d82f3e9/wrappers/python/examples/export_ply_example.py#L40)

MartyG-RealSense, 2022-10-31:
A RealSense team member advises at https://github.com/IntelRealSense/librealsense/issues/6677#issuecomment-648079238 that it is not possible to save to ASCII (non-binary) with export_to_ply.

save_to_ply has the option to set binary to false as you mention, though color export has not previously worked with it when its ignore_color parameter has been set to false.  It has been a couple of years since the Python script at https://github.com/IntelRealSense/librealsense/issues/7747#issuecomment-725346152 was last tested by a RealSense user though, so you could try it to see whether it can now export color.

1215232494, 2022-11-01:
Apologies for the delay in responding further. I read some answers( #9531) about you and understood the limitations of these methods. 
But I found a way to read binary ply files by using `plyfile.PlyData.read()`. It solves the problem that I want to read xyzrgb. 
Thanks for your help!  And I have a new problem, so I will close the case and start  a new one.

closed 2022-11-01 by 1215232494