realsenseai/librealsense #11047: WGL

aqshah20, 2022-10-31:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GLFW Driver Error: WGL: The driver does not appear to support OpenGL
Could not initialize offscreen context!

```

MartyG-RealSense, 2022-10-31:
Hi @aqshah20  I hope that the information at https://github.com/IntelRealSense/librealsense/issues/6976 will be helpful to resolving your problem.

aqshah20, 2022-10-31:
Thank you for your response
one thing more when I install and run nuitrack windows sdk after installed
nothing to execute anything

MartyG-RealSense, 2022-10-31:
Nuitrack SDK supports the RealSense models D415, D435, D435i D455 and L515.  Is your camera one of these models, please?

Also, do you have a Nuitrack licence subscription?  The trial version has a 3 minute time limit per session but can be used for an unlimited number of sessions.

If you have one of these cameras and you can use the RealSense SDK successfully but not Nuitrack SDK, please post a question on the Nuitrack support forum at the link below.

https://community.nuitrack.com/c/nuitrack-sdk/6

aqshah20, 2022-10-31:
I have this   D415 but issue is I can't even I followed the tutorial from
you github repo

aqshah20, 2022-10-31:
[image: image.png]

SDK is installed but not executing anything

MartyG-RealSense, 2022-10-31:
Please give me a link to the tutorial.

MartyG-RealSense, 2022-10-31:
Also, images cannot be posted to this support forum by email unfortunately.

aqshah20, 2022-10-31:
https://www.youtube.com/watch?v=F1Vx_pu9dbk
https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Basic.md
I have unity version 2021.3.11f1
D415 realsense

MartyG-RealSense, 2022-10-31:
Thank you.  This YouTube tutorial is published by the company that makes Nuitrack, so questions about it should be asked on the Nuitrack support forum, please.  Intel is unable to provide support on this forum for RealSense hardware or software products that are produced by companies other than Intel.

Please also ensure that you have activated the Nuitrack licence.

https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Install.md#license-activation

There is a text version of the basic Unity tutorial that you can check to confirm whether you have carried out every step of the setup process.

https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Basic.md

aqshah20, 2022-11-01:
getting ERROR: Sample failed. Exit code: 3 error
could you please guide me?

aqshah20, 2022-11-01:
nuitrack.ModuleNotInitializedException: NuitrackException
(ModuleNotInitializedException): Can't create DepthSensor module
C:\Program Files\Nuitrack\nuitrack\nuitrack/data//rgbSettings.config:
cannot open file

MartyG-RealSense, 2022-11-01:
Please post your question on the Nuitrack SDK support forum as Intel is unable to provide support advice for non-Intel products such as Nuitrack, unfortunately.

https://community.nuitrack.com/c/nuitrack-sdk/6

aqshah20, 2022-11-05:
Hello MartyG,
am facing a little issue can you please help me I posted in form but no
response of my two questions

current having this issue
Assets\Scripts\GameController.cs(65,38): error CS0103: The name
‘CurrentUserTracker’ does not exist in the current context
<https://community.nuitrack.com/t/assets-scripts-gamecontroller-cs-65-38-error-cs0103-the-name-currentusertracker-does-not-exist-in-the-current-context/3159>

MartyG-RealSense, 2022-11-05:
Whilst I cannot directly advise on this problem, there was a case on the Nuitrack forum with that error that was caused by a missing line at the top of the script.

https://community.nuitrack.com/t/solved-issue-with-unity-nuitrack-setup/2832

MartyG-RealSense, 2022-11-15:
Hi @aqshah20  Was the information in the comment above helpful to you, please?  Thanks!

MartyG-RealSense, 2022-11-15:
The information provided was:

Whilst I cannot directly advise on this problem, there was a case on the Nuitrack forum with that error that was caused by a missing line at the top of the script.

https://community.nuitrack.com/t/solved-issue-with-unity-nuitrack-setup/2832

closed 2022-11-20 by MartyG-RealSense