realsenseai/librealsense #11050: Kernel Patches Ubuntu

lantianhaohh, 2022-10-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| RealSense ROS                 |v2.3.2| 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.10.110                                        | 
| Platform                                 |rk3588  |
| SDK Version                            | LibRealSense v2.51.1                       | 

[image]

have followed the installed steps described at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md and am now stuck on running step 4, "./scripts/patch-realsense-ubuntu-lts.sh"
Is this script important and can I continue building realsense SDK?
Thanks

MartyG-RealSense, 2022-10-31:
Hi @lantianhaohh  Whilst you can build librealsense from source code without having patched the kernel, support for camera **hardware metadata** will not be included and frame timestamps based on the computer's **system time** clock will be used by librealsense instead.  This may not affect you negatively if your project does not require use of hardware timestamps.

lantianhaohh, 2022-10-31:
Hi @MartyG-RealSense , does the kernel patch support metadata and hardware timestamps for camera hardware?So if the kernel can be patched, will the camera's data communication in RK3588 be more stable?For the information printed above log, the kernel version adaptation problem, how to modify. Thank you

MartyG-RealSense, 2022-10-31:
Yes, when building from source code, patching the kernel adds support for hardware timestamps.  The kernel must be one that is supported by the patch script though.  With a source code build it is always preferable to patch the kernel if possible, except for an RSUSB Backend type source build where patching is not required.

I would strongly recommend an RSUSB build if you are unable to use a kernel supported by the RealSense SDK.

If you would prefer to explore kernel modification then a RealSense team member provides advice at https://github.com/IntelRealSense/librealsense/issues/3292#issuecomment-465876055

MartyG-RealSense, 2022-11-06:
Hi @lantianhaohh  Do you require further assistance with this case, please?  Thanks!

lantianhaohh, 2022-11-08:
@MartyG-RealSense  Hello, will your team come up with a kernel patch for the RK series?Thanks!

lantianhaohh, 2022-11-08:
@MartyG-RealSense Hello,thank you very much for your help and support

MartyG-RealSense, 2022-11-09:
Hi again @lantianhaohh  Do you have knowledge of how to compile a Linux kernel from source code please, like in the guide at the link below.

https://www.linux.com/topic/desktop/how-compile-linux-kernel-0/

lantianhaohh, 2022-11-09:
@MartyG-RealSense Hello, you just sent the link guide, I finished reading, watching the tutorial to do it, I compiled the kernel from source code should have no problem

MartyG-RealSense, 2022-11-09:
Thanks very much @lantianhaohh - if you require further assistance, please do let me know!

lantianhaohh, 2022-11-09:
Hi @MartyG-RealSense ,Although I don't know much about compiling the Linux kernel from source, it should be fine to watch the tutorial

MartyG-RealSense, 2022-11-10:
I have received further advice about your case from my colleagues.  They recommend contacting the vendor of your RK device for the source code of your 5.10 kernel, as according to their experience there may be differences between the RK source code of your device and the source code at the link that I provided above.

They added however that kernel 5.10 might already include most of the kernel patches, so you could certainly try compiling librealsense from source using the **native backend** installation method, which is used in the librealsense build when setting the build flag **-DFORCE_RSUSB_BACKEND=FALSE** instead of setting RSUSB to True.

**************

After the building of librealsense is complete, you could test it with the example program **rs-depth** which is not dependent on OpenGL graphics.  

https://github.com/IntelRealSense/librealsense/tree/master/examples/C/depth 

You could also test it with the example program **rs-hello-realsense** which similarly does not use OpenGL graphics.

https://github.com/IntelRealSense/librealsense/tree/master/examples/hello-realsense

*************

My colleagues also provided an **rs-capture** C++ script in the attached .txt file that could be built to check whether metadata can be read out.

[rs-capture.txt](https://github.com/IntelRealSense/librealsense/files/9979835/rs-capture.txt)

If the metadata is incomplete then they advise you to get the RK kernel source code and provide the list of kernel files below.

/drivers/iio/accel/hid-sensor-accel-3d.c
/drivers/iio/gyro/hid-sensor-gyro-3d.c
/drivers/iio/common/hid-sensors/hid-sensor-attributes.c 
/drivers/media/usb/uvc/uvc_ctrl.c 
/drivers/media/usb/uvc/uvc_driver.c 
/drivers/media/usb/uvc/uvc_video.c 
/drivers/media/usb/uvc/uvcvideo.h 
/drivers/media/v4l2-core/v4l2-ioctl.c 
/drivers/usb/host/xhci-ring.c 
/drivers/usb/host/xhci.c 
/drivers/usb/host/xhci.h 
/include/linux/hid-sensor-hub.h 
/include/linux/hid-sensor-ids.h 
/include/uapi/linux/videodev2.h

************

Thanks very much again for your patience, @lantianhaohh