realsenseai/librealsense #11053: How to set the short range of L515 with python code?

Drift-Of-Little-Forest, 2022-10-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I start the depth camera through python code, and the depth screen appears in the Max range mode, but my project needs to use the camera's short range mode. How to set the L515's short range mode through code in python? Is there an example code? Please help me

Drift-Of-Little-Forest, 2022-10-31:
[image]
！！How to use python code to implement this function on the SDK！！

MartyG-RealSense, 2022-10-31:
Hi @Drift-Of-Little-Forest  Here is Python code for setting the Short Range preset.

```
# Declare sensor object and set options
depth_sensor = profile.get_device().first_depth_sensor()
depth_sensor.set_option(rs.option.visual_preset, 5) # 5 is short range, 3 is low ambient light
```

https://github.com/IntelRealSense/librealsense/issues/8619 is an example of a full Python script that uses this code to define the preset.

Drift-Of-Little-Forest, 2022-11-01:
> Hi@Drift-of-Little-Forest以下是用于为 L515 相机型号设置短距离预设的 Python 代码。
> 
> ```
> # Declare sensor object and set options
> depth_sensor = profile.get_device().first_depth_sensor()
> depth_sensor.set_option(rs.option.visual_preset, 5) # 5 is short range, 3 is low ambient light
> ```
> 
> #8619是一个完整的 Python 脚本示例，该脚本使用此代码来定义预设。
Thank you very much for your help

Drift-Of-Little-Forest, 2022-11-01:
> Hi @Drift-Of-Little-Forest Here is Python code for setting the Short Range preset for the L515 camera model.
> 
> ```
> # Declare sensor object and set options
> depth_sensor = profile.get_device().first_depth_sensor()
> depth_sensor.set_option(rs.option.visual_preset, 5) # 5 is short range, 3 is low ambient light
> ```
> 
> #8619 is an example of a full Python script that uses this code to define the preset.

I want to know the difference between color alignment depth and depth alignment color, please help me

MartyG-RealSense, 2022-11-01:
With depth to color alignment, the size of the depth field of view (FOV) will be resized to the size of the color FOV.  When the color FOV is smaller then the depth FOV, this means that the outer edges of the depth image may be excluded from the aligned image.

With color to depth alignment, the color FOV is resized to match the depth FOV size.  When the color FOV is smaller, this means that the color data will stretch out to fit the depth FOV size and depth information at the image edges will not be excluded from the aligned image.

Drift-Of-Little-Forest, 2022-11-01:
> With depth to color alignment, the size of the depth field of view (FOV) will be resized to the size of the color FOV. When the color FOV is smaller then the depth FOV, this means that the outer edges of the depth image may be excluded from the aligned image.
> 
> With color to depth alignment, the color FOV is resized to match the depth FOV size. When the color FOV is smaller, this means that the color data will stretch out to fit the depth FOV size and depth information at the image edges will not be excluded from the aligned image.
> 
> The L515 camera model has a depth FOV of 70° × 55° and a color FOV of 70° × 43°.

Thanks for your help！

MartyG-RealSense, 2022-11-06:
Hi @Drift-Of-Little-Forest  Do you require further assistance with this case, please?  Thanks!

Drift-Of-Little-Forest, 2022-11-06:
> 嗨@Drift小森林请问您需要此案的进一步帮助吗？谢谢！

I want to know the specific concepts of different depth resolutions. For example, the higher the color resolution is, the more detailed the pixel points of the real image are. The higher the depth resolution is, the more accurate the depth distance of a point in the measured object is? If you have time, I hope you can help me again.

MartyG-RealSense, 2022-11-06:
Yes, the higher the color resolution the better the image quality, as a high resolution image is made up of many tiny pixels whilst a low resolution image is made up of a smaller number of large pixel 'blocks' that can look more blurred and muddy.

In regard to the depth image, accuracy does increase as resolution increases and decreases as resolution decreases.  Each camera model has a depth resolution at which accuracy will be optimal.  For the D415 camera model it is 1280x720 and for most other 400 Series models it is 848x480.  For the L515, **1024x768** will be the best depth resolution for accuracy.

MartyG-RealSense, 2022-11-06:
Yes, the higher the color resolution the better the image quality, as a high resolution image is made up of many tiny pixels whilst a low resolution image is made up of a smaller number of large pixel 'blocks' that can look more blurred and muddy.

In regard to the depth image, accuracy does increase as resolution increases and decreases as resolution decreases. Each camera model has a depth resolution at which accuracy will be optimal. For the D415 camera model it is 1280x720 and for most other 400 Series models it is 848x480. For the L515, 1024x768 will be the best depth resolution for accuracy.

MartyG-RealSense, 2022-11-06:
See above at https://github.com/IntelRealSense/librealsense/issues/11053#issuecomment-1304753468 for the answer  :)

Drift-Of-Little-Forest, 2022-11-06:
> 是的，颜色分辨率越高，图像质量越好，因为高分辨率图像由许多微小像素组成，而低分辨率图像由较少数量的大像素“块”组成，看起来更加模糊和模糊。
> 
> 关于深度图像，精度确实随着分辨率的增加而增加，随着分辨率的降低而降低。每个相机型号都有一个深度分辨率，在该分辨率下精度将是最佳的。对于 D415 相机型号，它是 1280x720，对于大多数其他 400 系列型号，它是 848x480。对于L515，**1024x768**将是精度的最佳深度分辨率。

Thank you very much for your help

MartyG-RealSense, 2022-11-12:
Hi @Drift-Of-Little-Forest  Do you require further assistance with this case please?  Thanks!

Drift-Of-Little-Forest, 2022-11-12:
Thank you very much for your help. I haven't encountered any new difficulties yet

?浮生未歇?
***@***.***

closed 2022-11-17 by MartyG-RealSense