realsenseai/librealsense #11054: Realsense D457 cable specifications for > 11 m length

Mat198, 2022-10-31:
Hello!
I'm looking forward to using Realsense D457, but I have doubts about its cable.

What is the maximum total resistance the cable can have? And the attenuation?

I want to try Dacar 302, as suggested here: https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1292320808, but it's unavailable in my country. There are other options besides that one?

MartyG-RealSense, 2022-11-01:
I discussed the subject of cable electronics further with the Intel RealSense team.  They stated that the D457 uses the GMSL protocol, so any GMSL cable should work.

Mat198, 2022-11-01:
Thanks for your reply!

I'm excited to test the new Realsense.

MartyG-RealSense, 2022-11-06:
Hi @Mat198  Do you require further assistance with this case, please?  Thanks!

Mat198, 2022-11-08:
I'm OK for now. I'm waiting for my D457 to arrive to test it.

MartyG-RealSense, 2022-11-08:
Thanks very much @Mat198 for the update.  Please do feel free to post further questions when you have the opportunity to test your D457.  Good luck!

MartyG-RealSense, 2022-11-18:
Hi @Mat198  Have you received your D457 and been able to test it, please?  Thanks!

Mat198, 2022-11-18:
Hi @MartyG-RealSense! I'm expecting to test it in the first half of January. I need the camera and the embedded computer with GMSL support to arrive at my company first.

MartyG-RealSense, 2022-11-18:
Okay, that's no problem at all.  If you wish you could close this issue now using the button under the comment writing window and re-open it in January or open a new issue.  Thanks again!

closed 2022-11-18 by Mat198