realsenseai/librealsense #11055: How to set the color value of color_frame?

1215232494, 2022-11-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |  Linux (Ubuntu 20)  |                                     | 
| Language                            |  python                          | 

### Issue Description
I got the rgb image from the [`color_frame.get_data()`](https://github.com/IntelRealSense/librealsense/blob/923253f645e867c75e3c6c30a84127f41d82f3e9/wrappers/python/examples/pyglet_pointcloud_viewer.py#L426), 
[image]

and I let it go through the neural network to get a mask
[image]

I can see the color of the mask on pyglet, Probably because [rgb image is used](https://github.com/IntelRealSense/librealsense/blob/923253f645e867c75e3c6c30a84127f41d82f3e9/wrappers/python/examples/pyglet_pointcloud_viewer.py#L454)
[image]
 
But when I [export the ply file](https://github.com/IntelRealSense/librealsense/blob/923253f645e867c75e3c6c30a84127f41d82f3e9/wrappers/python/examples/pyglet_pointcloud_viewer.py#L491), I get the color of the original image instead of the mask, because the [pointcloud uses color_ frame](https://github.com/IntelRealSense/librealsense/blob/923253f645e867c75e3c6c30a84127f41d82f3e9/wrappers/python/examples/pyglet_pointcloud_viewer.py#L437)
[image]

So, how do I assign the mask to color_frame so that the ply file is the color of the mask? Or whether there are other ways to achieve this effect?

MartyG-RealSense, 2022-11-01:
Hi, the only Python pointcloud reference that I could find about defining a mask in a program that uses the **export_to_ply** 
 ply export instruction is in the SDK example program **opencv_pointcloud_viewer.py**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L255-L258

1215232494, 2022-11-02:
It seems that we can't get the ply file of mask color. I tried to add `m = m & mask` after [these lines](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L255-L258), but the displayed content did not change. What should I do?
Alternatively, if I change the color to mask before this line, I will get the mask color display. But when I press "E" to save the ply file, I get the original color instead of the mask color

MartyG-RealSense, 2022-11-02:
Does the mask get saved to ply if you incorporate your **mask** change into the 'm' instruction?

`m = im & jm & mask`

Another approach may to to give the original 'm; a different name and then apply your mask to the output of the renamed 'm'.  for example, if you renamed 'm' to 'm_original':

```
im = (i >= 0) & (i < h)
jm = (j >= 0) & (j < w)
m_original = im & jm
m = m_original & mask
```

MartyG-RealSense, 2022-11-09:
Do you require further assistance with this case, please?  Thanks!

1215232494, 2022-11-12:
Apologies for the delay in responding further.The suggestions you provided do not solve the problem. I have no other solutions.

MartyG-RealSense, 2022-11-12:
I do not have any further references to suggest either other than looking at how masking is handled with OpenCV code in the **rs-grabcuts** example program.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/grabcuts

MartyG-RealSense, 2022-11-21:
Was the advice in the comment above helpful, please?  Thanks!

closed 2022-11-27 by MartyG-RealSense