realsenseai/librealsense #11056: How to delete points in the point cloud according to the value of color_image?

1215232494, 2022-11-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |  Linux (Ubuntu 20)  | 
| Language                            | python }                          | 

### Issue Description
I got the rgb image from the [`color_frame.get_data()`](https://github.com/IntelRealSense/librealsense/blob/923253f645e867c75e3c6c30a84127f41d82f3e9/wrappers/python/examples/pyglet_pointcloud_viewer.py#L426), 
[image]

and I let it go through the neural network to get a target area
[image]

I add the following code after [this line](https://github.com/IntelRealSense/librealsense/blob/923253f645e867c75e3c6c30a84127f41d82f3e9/wrappers/python/examples/pyglet_pointcloud_viewer.py#L456)
`verts[color_image==[0,0,0]]=0.`

and get results
[image]

This is obviously a problem. A small part of the target area on the left is missing, while other non target areas are retained.  And I must ensure that the size of color_frame and depth_frame are consistent before I can delete the pointcloud.  
Is there any other way to solve these problems?

MartyG-RealSense, 2022-11-01:
The missing detail on the left side of the box is likely due to a phenemenon called the **Invalid Depth Band** that is described on page 79 of the data sheet document for the 400 Series cameras.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

*******

The depth data generated by the D4 Vision Processor utilizes the left imager as the reference for the stereo matching algorithm, resulting in a non-overlapped region in the camera’s field of view.  This non-overlapped region (at the left edge of the frame) contains no depth data.  The invalid depth data band decreases as the distance from the camera / module to the scene increases.

*******

If the missing detail is due to the Invalid Depth Band then the data may return if the camera is moved further away from the box so that the width of the Band reduces.

1215232494, 2022-11-01:
Oh yes. I solved the problem of missing the left side by moving the camera position
[image]

The question now is why other non target areas are retained

MartyG-RealSense, 2022-11-01:
A depth image is constructed from the left and right sensor images.  So whilst the left edge of the depth image may be cut off due to the area of non-overlap between the left and right sensors, the areas at the top and bottom of the image should still be displayed so long as the sensors can see those areas of the scene.

1215232494, 2022-11-01:
But my mask does not include these areas. Normally, the point clouds in the surrounding areas will be deleted, just like the point clouds far away from the target area.
The point cloud before deletion is as follows
[image]

1215232494, 2022-11-01:
Thanks for your patience, I think I probably understand what you mean. When I move the camera to the right, the reserved area comes to the left. 
[image]

Is there any way to avoid this situation, because my application scenario needs to solve this problem, or solving another question #11055 can also replace this problem

MartyG-RealSense, 2022-11-01:
Does the image improve if you change the depth unit scale from its default value of 0.001 to **0.01** using the Python code at https://github.com/IntelRealSense/librealsense/issues/10976#issuecomment-1271236193

1215232494, 2022-11-02:
I tried, but there was no obvious improvement effect, and the error of depth data became large.

0.001
[image]

to  0.01
[image]

1215232494, 2022-11-02:
Hi @MartyG-RealSense  , I made a low-level mistake. I should align the mask with the depth map before deleting the point cloud. I corrected and got the result I wanted. Thanks for your help!
[image]

closed 2022-11-02 by 1215232494

MartyG-RealSense, 2022-11-02:
You are vey welcome.  It's excellent news that you achieved a solution.  Thanks very much for the update!