realsenseai/librealsense #11057: L515-How to extract poses and the intrinsics of the camera

Ceciliammm, 2022-11-01:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1371405|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003cbb76779|
|**Firmware Update Id**|0003cbb76779|

Please provide a description of the problem

MartyG-RealSense, 2022-11-01:
Hi @Ceciliammm  You can retrieve the L515 camera's intrinsics and extrinsics (which include translation and rotation) with the rs-enumerate-devices tool using the instructions at https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-615365409

Ceciliammm, 2022-11-01:
> Hi @Ceciliammm You can retrieve the L515 camera's intrinsics and the extrinsics (which include translation and rotation) with the pre-made Windows version of the RealSense SDK's **rs-enumerate-devices** tool using the instructions at [#3569 (comment)](https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-615365409)

but can I extract the intrinsics and extrinsics from the recorded BAG file of the camera L515?

MartyG-RealSense, 2022-11-01:
You can set a bag file as the data source for a script instead of a live L515 camera using the SDK instruction **cfg.enable_device_from_file** and the script will behave as though a live camera is is being used with it.

If you are able to use Python scripting then https://github.com/IntelRealSense/librealsense/issues/7309#issuecomment-691366761 has a Python script for getting intrinsics from a bag.

https://github.com/IntelRealSense/librealsense/issues/10457 has Python scripting for accessing extrinsics with the **get_extrinsics_to** instruction.

Ceciliammm, 2022-11-01:
> You can set a bag file as the data source for a script instead of a live L515 camera using the SDK instruction **cfg.enable_device_from_file** and the script will behave as though a live camera is is being used with it.
> 
> If you are able to use Python scripting then [#2869 (comment)](https://github.com/IntelRealSense/librealsense/issues/2869#issuecomment-445771264) has a Python script for getting intrinsics with the **get_intrinsics** instruction.
> 
> #10457 has Python scripting for accessing extrinsics with the **get_extrinsics_to** instruction.

but when I add a bag file as the source, when I run the script that extract intrinsics, something goes wrong with the following result:
[image]

MartyG-RealSense, 2022-11-01:
In the script at https://github.com/IntelRealSense/librealsense/issues/10457
they had not included **cfg.enable_device_from_file** in the script to tell the program to use the bag as the data source instead of a live camera.

The enable_device_from_file instruction should also be placed **before** the pipeline start line, like in the example code at the link below.

https://stackoverflow.com/questions/58482414/frame-didnt-arrived-within-5000-while-reading-bag-file-pyrealsense2

Can you confirm please that your extrinsics-accessing code has the enable_device_from_file instruction and that the instruction is placed before the pipe start line in your script, please?

Ceciliammm, 2022-11-02:
> In the script at #10457 they had not included **cfg.enable_device_from_file** in the script to tell the program to use the bag as the data source instead of a live camera.
>
> The [...]

My problem has been solved. Thank you very much!

closed 2022-11-02 by Ceciliammm