realsenseai/librealsense #11059: L515: RuntimeError: Frame didn't arrive within 5000

Ceciliammm, 2022-11-02:
`|` | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1371405|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003cbb76779|
|**Firmware Update Id**|0003cbb76779|

Please provide a description of the problem
import pyrealsense2 as rs
import numpy as np

def initialize_camera():
    conf = rs.config()
    rs.config.enable_device_from_file(conf, "F:/NeRF_Project_dataset/20221031_160514.bag", repeat_playback=False)
    p = rs.pipeline()
    # conf.enable_stream(rs.stream.accel)
    # conf.enable_stream(rs.stream.gyro)
    p.start(conf)
    return p

def gyro_data(gyro):
    return np.asarray([gyro.x, gyro.y, gyro.z])

def accel_data(accel):
    return np.asarray([accel.x, accel.y, accel.z])

p = initialize_camera()
accel_total = []
gyro_total = []
try:
    while True:
        f = p.wait_for_frames()
        # print(f)
        accel = accel_data(f[3].as_motion_frame().get_motion_data())
        accel_total.append(accel)
        gyro = gyro_data(f[4].as_motion_frame().get_motion_data())
        gyro_total.append(gyro)
        # print(len(accel_total), len(gyro_total))
        # print("accelerometer: ", accel)
        # print("gyro: ", gyro)

finally:

    p.stop()
    print(len(accel_total), len(gyro_total))

MartyG-RealSense, 2022-11-02:
Hi @Ceciliammm  You are reading the data from a recorded bag file instead of a live L515 camera, so the camera will not be the cause of this error.

Can you confirm please that the bag file contains accel and gyro data?  As a bag file is being used as the data source, the script can only retrieve information that is stored in the bag.

Also, you should define a custom stream configuration that matches the configuration of the streams that are stored in the bag.  The default Hz speed value on L515 for both the accel and gyro is **100** Hz.

[image]
 
Therefore, please try the following configuration lines in your script without the # comment-out symbol in front of them.

```
conf.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)
conf.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 100)
```

*********

There was another Python case on L515 at https://github.com/IntelRealSense/librealsense/issues/6896 where the **RuntimeError: Frame didn't arrive within 5000** error was occurring.  A RealSense team member in that case recommends at https://github.com/IntelRealSense/librealsense/issues/6896#issuecomment-665274907 to check if there is a valid accel or gyro stream with an 'if' statement before printing the values.

They also recommend to identify **dynamically** whether the stream is of the accel or gyro type instead of using fixed index numbers ('3' and '4' in your script), otherwise the stream output could be reversed (gyro values printed for accel and accel values printed for gyro).

Ceciliammm, 2022-11-02:
> Hi @Ceciliammm You are reading the data from a recorded bag file instead of a live L515 camera, so the camera will not be the cause of this error.
> 
> Can you confirm please that the bag file contains accel and gyro data? As a bag file is being used as the data source, the script can only retrieve information that is stored in the bag.
> 
> Also, you should define a custom stream configuration that matches the configuration of the streams that are stored in the bag. The default Hz speed value on L515 for both the accel and gyro is **100** Hz.
> 
> [image]
> 
> Therefore, please try the following configuration lines in your script without the # comment-out symbol in front of them.
> 
> ```
> conf.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)
> conf.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 100)
> ```
> 
> There was another Python case on L515 at #6896 where the **RuntimeError: Frame didn't arrive within 5000** error was occurring. A RealSense team member in that case recommends at [#6896 (comment)](https://github.com/IntelRealSense/librealsense/issues/6896#issuecomment-665274907) to check if there is a valid accel or gyro stream with an 'if' statement before printing the values.
> 
> They also recommend to identify **dynamically** whether the stream is of the accel or gyro type instead of using fixed index numbers ('3' and '4' in your script), otherwise the stream output could be reversed (gyro values printed for accel and accel values printed for gyro).
Thank you for your help, but I still have the following problems:
1. My bag file contains gyro and accel data. As shown below:
[image]
2. After add these two lines, my project still has the same problem. [RuntimeError: Frame didn't arrive within 5000]
conf.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)
conf.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 100)
3. If the Accel and Gyro wiould exchange, how can I match them correctly?

Ceciliammm, 2022-11-02:
> [#4018 (comment)](https://github.com/IntelRealSense/librealsense/issues/4018#issuecomment-520827544) has a pyrealsense2 code reference for setting the stream type dynamically.
> 
> I also note that you are using an **append** instruction - **accel_total.append(accel)** and **gyro_total.append(gyro)**
> 
> Using the append instruction in pyrealsense2 scripts can cause problems, as it can cause the script to receive the first 15 frames and then stop at the 16th. This may be causing the program to stop receiving new frames and 'time out', resulting in the 'Frame didn't arrive within 5000' error. Using a **Keep()** instruction to store frames in memory is a workaround for this problem, as described at [#8652 (comment)](https://github.com/IntelRealSense/librealsense/issues/8652#issuecomment-804993692)

when I use accel.keep(), then "AttributeError: 'numpy.ndarray' object has no attribute 'keep'"
In addition, the following codes only read the accel data, how about the gyro data?
        for frame in f:
            if frame.is_motion_frame():
                accel = accel_data(frame.as_motion_frame().get_motion_data())

MartyG-RealSense, 2022-11-02:
Instead of using accel.keep() please try inserting the keep instruction after the wait_for_frames line and using 'f' instead of 'accel'.

```
f = p.wait_for_frames()
f.keep()
```

**********

My understanding from past references about dynamic assigning of accel and gyro in Python is that you simply create separate entries for accel and gyro under the **if frame.is_motion_frame():** statement.

```
for frame in f:
if frame.is_motion_frame():
accel = accel_data(frame.as_motion_frame().get_motion_data())
gyro = gyro_data(frame.as_motion_frame().get_motion_data())
```

The variable names **accel_data** and **gyro_data** used by the above lines were defined earlier in your script.

[image]

Ceciliammm, 2022-11-03:
> 

My bag file contains gyro and accel data.
How can I define a custom stream configuration?

> Instead of using accel.keep() please try inserting the keep instruction after the wait_for_frames line and using 'f' instead of 'accel'.
>
> ```
> f = p.wait_for_frames()
> f.keep()
> ```
>
> My understanding [...]

1. About the f.keep():
 After I change the codes to 
f = p.wait_for_frames()
        f.keep()
My project still has the same error "RuntimeError: Frame didn't arrive within 5000"
2. If I store the gyro data the same as accel's then their data will be the same. But the gyro data and the accel data are different.
[image]

The total codes of my project is shown below:
```
`import pyrealsense2 as rs
import numpy as np

def initialize_camera():
    conf = rs.config()
    rs.config.enable_device_from_file(conf, "F:/NeRF_Project_dataset/20221031_160514.bag", repeat_playback=False)
    p = rs.pipeline()
    conf.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)
    conf.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 100)
    p.start(conf)
    return p

def gyro_data(gyro):
    return np.asarray([gyro.x, gyro.y, gyro.z])

def accel_data(accel):
    return np.asarray([accel.x, accel.y, accel.z])

p = initialize_camera()
# accel_total = []
# gyro_total = []
try:
    while True:
        f = p.wait_for_frames()
        f.keep()

        # print(f)
        for frame in f:
            if frame.is_motion_frame():
                accel = accel_data(frame.as_motion_frame().get_motion_data())
                gyro = gyro_data(frame.as_motion_frame().get_motion_data())

                # gyro = gyro_data(frame.as_motion_frame().get_motion_data())
        # accel = accel_data(f[0].as_motion_frame().get_motion_data())
        # # accel_total.append(accel)
        # gyro = gyro_data(f[1].as_motion_frame().get_motion_data())
        # gyro_total.append(gyro)
        # print(len(accel_total), len(gyro_total))
        print("accelerometer: ", accel)
        print("gyro: ", gyro)

finally:

    p.stop()
    # print(len(accel_total), len(gyro_total))`
```

MartyG-RealSense, 2022-11-03:
You have already defined a custom stream configuration for accel and gyro by creating **conf** statements and placing the word 'conf' in the brackets of the pipeline start instruction so that the script knows to use the custom configuration.  

```
conf = rs.config()

p = rs.pipeline()
conf.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)
conf.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 100)
p.start(conf)
```

Please try changing wait_for_frames() to **try_wait_for_frames()** because if a time-out occurs then this should cause a 'false' status to be returned instead of the program exiting with a time-out error.

MartyG-RealSense, 2022-11-03:
Okay, let's comment out the **f.keep()** line for the moment and see what happens now that try_wait_for_frames is being used.

Ceciliammm, 2022-11-03:
> Okay, let's comment out the **f.keep()** line for the moment and see what happens now that try_wait_for_frames is being used.

It can not work because the line "frame.is_motion_frame():" cannot work.
[image]

MartyG-RealSense, 2022-11-03:
Let's try going back to the index-based approach for the moment just to get the output working even if there is a possibility that the accel and gyro might be reversed, and use **try_wait_for_frames()** too to try to avoid the time-out error.

```
for frame in f:
if frame.is_motion_frame():
accel = accel_data(f[3].as_motion_frame().get_motion_data())
gyro = gyro_data(f[4].as_motion_frame().get_motion_data())
```

Ceciliammm, 2022-11-03:
> Let's try going back to the index-based approach for the moment just to get the output working even if there is a possibility that the accel and gyro might be reversed, and use [...]

Don't work. Since when "try_wait_for_frames()" is used, the type of f becomes "bool",  'bool' object has no attribute 'as_motion_frame'.
[image]

Ceciliammm, 2022-11-03:
> Ok, that is understandable since a bool status ('false' when timeout occurs) would be being returned.
> 
> I'm not sure that **return p** on the line after pipe start is needed, so is there any change if that line is commented out?

p is used to initialize and stop the camera, if I comment out return p, I think the total project cannot work.
My codes are shown below:
```
import pyrealsense2 as rs
import numpy as np
import os

def initialize_camera():
    conf = rs.config()
    rs.config.enable_device_from_file(conf, "F:/NeRF_Project_dataset/20221031_160514.bag", repeat_playback=False)
    p = rs.pipeline()
    conf.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)
    conf.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 100)
    p.start(conf)
    return p

def gyro_data(gyro):
    return np.asarray([gyro.x, gyro.y, gyro.z])

def accel_data(accel):
    return np.asarray([accel.x, accel.y, accel.z])

p = initialize_camera()
i = 1
basedir = r"C:\Users\admin\Desktop\Cecilia"
accel_path = os.path.join(basedir, 'accel.txt')
gyro_path = os.path.join(basedir, 'gyro.txt')
try:
    with open(accel_path, "w") as file_a, open(gyro_path, "w") as file_g:
        while True:
            f = p.try_wait_for_frames()
            # f.keep()

            # print(f)
            for frame in f:
                # if frame.is_motion_frame():
                accel = accel_data(f[0].as_motion_frame().get_motion_data())
                gyro = gyro_data(f[1].as_motion_frame().get_motion_data())

                print("accelerometer: ", accel)
                print("gyro: ", gyro)

                file_a.write(str(accel))
                file_a.write('\n')

                file_g.write(str(gyro))
                file_g.write('\n')

            i = i+1
            print(i)

finally:

    p.stop()
````

MartyG-RealSense, 2022-11-03:
I understand that 'p' is the pipeline definition that is used to start and stop the camera.

I would think though that if **return p** was commented out then the script would run through once and then be stopped by **p.stop()** before it could loop around again for another run-through.

Thanks very much for your patience during this case!

Ceciliammm, 2022-11-03:
> I understand that 'p' is the pipeline definition that is used to start and stop the camera.
> 
> I would think though that if **return p** was commented out then the script would run through once and then be stopped by **p.stop()** at the end of the script before it could loop around again for another run-through.
> 
> Thanks very much for your patience during this case!

I'm so sorry that I don't know how to modify my codes to satisfy your requirements to test the necessity of return p. 
Since if I comment out return p, then p.try_wait_for_frames(), the "try..finally.." structure as well as p.stop should be changed. 
To be specifically, I don't know how to change p.try_wait_for_frames().

Thank you for your prompt reply.

MartyG-RealSense, 2022-11-03:
It's no problem at all, don't worry!

What I am suggesting is:

1.  Comment out **return p**

[image]

2.  Change try_wait_for_frames() back to **wait_for_frames()**

[image]

The aim of this is to see if 'return p' is what is causing the time-out error.

Ceciliammm, 2022-11-03:
> It's no problem at all, don't worry!
> 
> What I am suggesting is:
> 
> 1. Comment out **return p**
> 
> [image]
> 
> 2. Change try_wait_for_frames() back to **wait_for_frames()**
> 
> [image]
> 
> The aim of this is to see if 'return p' is what is causing the time-out error.

Bad news. 'return p' is not the factor causing the time-out error.

[image]

And the modified codes are shown here:

```
import pyrealsense2 as rs
import numpy as np
import os
def gyro_data(gyro):
    return np.asarray([gyro.x, gyro.y, gyro.z])

def accel_data(accel):
    return np.asarray([accel.x, accel.y, accel.z])

# p = initialize_camera()
# initialize_camera()
conf = rs.config()
rs.config.enable_device_from_file(conf, "F:/NeRF_Project_dataset/20221031_160514.bag", repeat_playback=False)
p = rs.pipeline()
conf.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)
conf.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 100)
p.start(conf)

i = 1
basedir = r"C:\Users\admin\Desktop\Cecilia"
accel_path = os.path.join(basedir, 'accel.txt')
gyro_path = os.path.join(basedir, 'gyro.txt')
try:
    with open(accel_path, "w") as file_a, open(gyro_path, "w") as file_g:
        while True:
            f = p.wait_for_frames()
            # f.keep()

            # print(f)
            for frame in f:
                # if frame.is_motion_frame():
                accel = accel_data(f[0].as_motion_frame().get_motion_data())
                gyro = gyro_data(f[1].as_motion_frame().get_motion_data())

                print("accelerometer: ", accel)
                print("gyro: ", gyro)

                file_a.write(str(accel))
                file_a.write('\n')

                file_g.write(str(gyro))
                file_g.write('\n')

            i = i+1
            print(i)

finally:

    p.stop()
```

MartyG-RealSense, 2022-11-03:
How long in minutes is the bag file recording of Accel and gyro data?

Have you tested that the file can be successfully played back by dragging and dropping the file into the center panel of the RealSense Viewer tool?

If the Viewer can play back the bag, does the 'Frame didn't arrive' error still occur if you remove **repeat_playback=False** so that the bag can loop playback?

Ceciliammm, 2022-11-04:
> How long in minutes is the bag file recording of accel and gyro data?
>
> Have you tested that the file can be successfully played back by dragging and dropping the file into the center panel of the [...]

1. 1:23 mins
2. It can be played back
3. If I change repeat_back from "False" to "True", the codes would run continuouslt without stop

Ceciliammm, 2022-11-04:
> How long in minutes is the bag file recording of accel and gyro data?
>
> Have you tested that the file can be successfully played back by dragging and dropping the file into the center panel of the [...]

Oh I know, maybe the RuntimeError is caused by the end of my recorded bag file.
And I have another question, how to change the imu data into camera's poses, in other words, extrinsics?

MartyG-RealSense, 2022-11-04:
Regarding the questions about different numbers of messages and about extrinsics, I have replied to your other issues at https://github.com/IntelRealSense/librealsense/issues/11066 and https://github.com/IntelRealSense/librealsense/issues/11067

Ceciliammm, 2022-11-07:
> Regarding the questions about different numbers of messages and about extrinsics, I have replied to your other issues at #11066 and #11067

Ok, thank you very much.

closed 2022-11-07 by Ceciliammm