realsenseai/librealsense #11065: L515-How to use IMU data (stored in bag file) to calculate camera extrinsics (in OpenGL or OpenCV)?

Ceciliammm, 2022-11-04:
[image]

MartyG-RealSense, 2022-11-04:
Hi @Ceciliammm  There is an example program for the T265 Tracking Camera model that retrieves the extrinsics and uses them in OpenCV.  That code may be adaptable for your L515 project.

**Retrieve the extrinsics**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py#L43-L61

**Then use get_extrinsics to get the relative **(R, T)** extrinsics between left and right**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py#L156-L157

***********

A project at https://github.com/IntelRealSense/librealsense/issues/10252 for the D415 camera model may be a more directly relevant reference for your project as it uses the same approach to get the (R,T) extrinsics from a **bag file** with OpenCV using get_extrinsics.  That project is more complex than yours though as they are attempting to get extrinsics from two bag files instead of one.

Ceciliammm, 2022-11-04:
> Hi @Ceciliammm There is an example program for the T265 Tracking Camera model that retrieves the extrinsics and uses them in OpenCV. That code may be adaptable for your L515 project.
> 
> **Retrieve the extrinsics** https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py#L43-L61
> 
> **Then use get_extrinsics to get the relative **(R, T)** rotation and translation extrinsics between left and right** https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py#L156-L157
> 
> A project at #10252 for the D415 camera model may be a more directly relevant reference for your project as it uses the same approach to get the (R,T) extrinsics from a RealSense **bag file** with OpenCV using get_extrinsics.
> 
> The above projects do not describe how to use **IMU** extrinsics in OpenCV but are the closest available references to what you are aiming to achieve.

Actually, I want to find the rotation matrix and the translation vector of the whole L515 camera (the two matrices form the camera's extrinsic matrix). Not the extrinsics between different sensors of the L515 camera.

MartyG-RealSense, 2022-11-04:
It sounds as though you want the **pose** (position and rotation) of the entire camera device rather than the extrinsics (which describe the relationship between two sensors on the same camera).

If you do want the pose then the L515 does not have built-in support for finding the camera's own pose, unfortunately (only the RealSense T265 Tracking Camera model does).

Ceciliammm, 2022-11-07:
> It sounds as though you want the **pose** (position and rotation) of the entire camera device rather than the extrinsics (which describe the relationship between two sensors on the same camera).
>
> If [...]

Ok, thank you very much.

closed 2022-11-07 by Ceciliammm