realsenseai/librealsense #11068: 3D mode not working

RupeshShirose, 2022-11-04:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D415|
|**Serial Number**|839512060088|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|844213023311|
|**Firmware Update Id**|844213023311|

Please provide a description of the problem
[image]

MartyG-RealSense, 2022-11-04:
Hi @RupeshShirose  If you have no pointcloud in 3D mode then disabling **GLSL** mode in the RealSense Viewer's settings interface using the instructions at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023 may resolve the issue.

MartyG-RealSense, 2022-11-17:
Case closed due to no further comments received.

closed 2022-11-17 by MartyG-RealSense

RupeshShirose, 2022-11-19:
Yeah, it worked. Thank you for the support