realsenseai/librealsense #11069: Doubt about L515 calibration

quizzy5889, 2022-11-04:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515  | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu 20.04 LTS  | 
| Kernel Version (Linux Only)    |  5.4.x  | 
| Platform                                 | PC  |
| SDK Version                            |   2.50.0  | 
| Language                            | python  | 
| Segment			|  3D Reconstruction | 

### Issue Description
Hello everyone. 
I am opening this topic because I have been struggling with this problem for a while now.

The idea I am trying is to try to reconstruct the 3D shape of an object by moving a L515 Lidar around it. I am mainly interested in reconstructing its frontal view, so fortunately I can use fiducial markers in the background to retrieve the relative transformation from the origin. These fiducial markers are Aruco markers placed in a Charuco board and the origin is the bottom left corner of the board. The board is hung on the wall.

So I did this: I acquired for each position aligned depth and rgb images. Using the distortion parameters I corrected the RGB image. I used OpenCV to extract the position of the markers in the RGB image. Don't worry about the tape you see on the picture. The calibrator is perfectly planar.

[image]

Exploiting rgb intrinsics I computed the relative transformation between RGB sensor and the board. If I project a reference system onto this image an error can already be seen.

[image]

The point is that if I transform the point cloud using the transformation I am unable to align the point cloud correctly. I transform the point cloud so that the expected final origin of the reference system corresponds to the corner of the board.

[image]

The error is approx 2 cm, but it's enough to make the final reconstruction inaccurate.

My guess is that it is a calibration problem, but I'm just guessing. Does anyone have ideas on how to solve it?

MartyG-RealSense, 2022-11-04:
Hi @quizzy5889  I recommend downloading the **User Guide** for the L515 camera model from the link below to find excellent advice on improving the results of the camera, such as trying different camera configuration **Visual Presets**.  Page 10 of the guide provides details of the different presets and their effects.

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

Changing the Visual Preset will enable you to configure the camera to be best suited for the environment that it is being used in.  For example, 'No Ambient Light' takes account of use in locations where there is no natural light, whilst 'Low Ambient Light' takes account of the presence of natural sunlight in the scene such as light entering through windows.

A Python script at https://github.com/IntelRealSense/librealsense/issues/8619 provides a demonstration of setting the preset using the SDK instruction **rs.option.visual_preset**

quizzy5889, 2022-11-05:
Thanks @MartyG-RealSense for your reply.
I just want to be sure that I got what you mean.

First of all, did you find any theoretical problem in the process I explained? Since rgb and depth are aligned I think that’s perfectly ok, but I wanted a double check from an Intel expert.

Secondly, you’re talking about presets. Does this mean that the Lidar outputs different pointclouds in term of points position depending on the preset?

Anyway I will try for sure what you suggest, but to be honest the pointcloud seems quite accurate, especially on the white part of the board.

Thanks again for your time

MartyG-RealSense, 2022-11-05:
My understanding is that a Charuco board provides better accuracy at the corners than an Aruco board, so your choice of Charuco is a good one for this particular application.

As mentioned in the User Guide for the L515, when observing a smooth object the accuracy may be better if the camera is tilted at a diagonal downwards angle when observing the surface instead of pointed straight downwards.  It looks as though you are already using a diagonal observation angle in the third of the images above though, so that is fine too.

Changing the preset can alter how accurately the camera can read depth information from a scene when certain light conditions are present or for certain materials that the objects being observed are made from.

An example would be a transparent drinks bottle.  The default settings of the L515 may look straight through the transparent bottle and only read the depth of the solid objects behind the bottle.  Setting the 'Short Range' preset reduces the values of the Laser Power and the Receiver Gain settings, making it easier for the camera to read depth values from the bottle.

If you are performing depth to color alignment with Python code in the RealSense SDK then there may be inaccuracy if you are using **depth** intrinsics.  When aligning depth to color, the **color** intrinsics should be used.

MartyG-RealSense, 2022-11-16:
Case closed due to no further comments received.

closed 2022-11-16 by MartyG-RealSense

quizzy5889, 2022-11-18:
Sorry @MartyG-RealSense for this late reply but I had a hard time trying to make everything work.

I tried different presets as you suggested but noone solved or improved in terms of final reconstruction accuracy.

From my understanding this could be something related to a slight misalignment between the rgb frame and the depth frame. 

At the end I decided to refine each and every scan pose using icp (basically I built a posegraph). The thing is that icp is not usable in every condition and it’s also time consuming…

MartyG-RealSense, 2022-11-19:
Hi @quizzy5889  As mentioned above, if you are performing depth to color alignment with Python code in the RealSense SDK then there may be inaccuracy in the alignment if you are using depth intrinsics.  When aligning depth to color, the **color** intrinsics should be used.

quizzy5889, 2022-11-19:
I confirm that I am using color intrinsics

MartyG-RealSense, 2022-11-20:
Does it make any difference to accuracy if you rotate the charuco board 90 degrees so that it is mounted vertical (portrait) instead of landscape (horizontal).