realsenseai/librealsense #11073: rs-callback gives no output

mathislm, 2022-11-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Linux (Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra | 
| Platform                                 | NVIDIA Jetson AGX Xavier  |
| SDK Version                            |  2.49.0                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
Hello,

When running `rs-callback`, there is no output given and we get the error **USBDEVFS_CLEAR_HALT** from `dmesg`. The `realsense-viewer` and other code samples like `rs-align` work fine, so I think the problem is related to using the callback method. However, we need to use callbacks in our application.
Do you have a solution for this problem ?

Best,
Mathis

MartyG-RealSense, 2022-11-07:
Hi @mathislm  **USBDEVFS_CLEAR_HALT** is a rarely reported error.  It has happened with other applications such as realsense-viewer and so may not be a callback-specific issue, but rather an issue that happens with callbacks on your particular Jetson computer.

An example of when it has occurred is when CPU percentage utilization is high (i.e the computer's CPU is over-burdened with processing).  Checking the CPU percentage usage with Ubuntu's **htop** system monitoring tool whilst rs-callback is running will show whether the CPU is overloaded with work.  Based on previous reports of the error, there may be no output because the program is freezing.

Reducing the resolution helped to resolve the error in some cases, though this is not an option in the rs-callback script.

A common factor between some of the reports was that the librealsense SDK was built with the **RSUSB backend* installation method.  What method did you use to install librealsense on your Jetson, please?

mathislm, 2022-11-07:
Hi @MartyG-RealSense, thanks for the answer, here is an example of the htop result, I don't think it is caused by resource shortage (I hope because we plan to use up to 6 cameras on this platform). 
[image]

This is the arguments we use to build librealsense : 
`cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=$(which python3) -DCMAKE_BUILD_TYPE=Release -DBUILD_NETWORK_DEVICE=ON -DFORCE_RSUSB_BACKEND=ON`

RSUSB is indeed on, however I don't remember why I enabled it... do you know what it is useful for ?

MartyG-RealSense, 2022-11-07:
Yes, your CPU percentage usage looks low.

**-DFORCE_RSUSB_BACKEND=ON** bypasses the Linux kernel, making a librealsense build not dependent on Linux versions or kernel versions and not needing to be kernel patched before librealsense is built.  It is not suited to use with multiple cameras though.  Setting RSUSB to **Off** would be ideal for a multicam project.

Including **-DBUILD_NETWORK_DEVICE=ON** will only be necessary if you plan to use ethernet networking to communicate with cameras.  It is not likely related to your USBDEVFS_CLEAR_HALT problem though.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

If you plan to use six cameras then I would recommend attaching the cameras to a mains electricity powered USB 3 hub if that is an option for your project in order to provide the cameras with stable power.

On an Nvidia Jetson build of librealsense, the CMake flag **-DBUILD_WITH_CUDA=true** will usually be included in the build instruction to enable support for CUDA graphics acceleration via the Nvidia graphics GPU on the Jetson board.

There are dedicated installation instructions for Jetson at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

mathislm, 2022-11-07:
I reinstalled the lib with RSUSB turned off and the code sample rs-callback works now! Thanks for the help. However, after testing our callback app, now both depth and rgb are frozen and here is what I read in `dmseg` :
`uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).`

Any idea what might be the cause ?

MartyG-RealSense, 2022-11-07:
As it is a **uvcvideo** error, it likely relates to the kernel.  A non-RSUSB build from source code (known as the 'V4L backend' or 'native backend') needs to have a patch applied to the kernel.  Information about patching for Jetson is in the **Build and Patch Kernel Modules for Jetson L4T** section at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

Also remember when writing your own callback app to insert the word **callback** into the brackets of the pipeline start instruction so that librealsense knows that it is a callback script.  For example, from the rs-callback script:

**rs2::pipeline_profile profiles = pipe.start(callback);**

mathislm, 2022-11-14:
Hello Marty,

I apologize for the very long delay, we are having trouble re-flashing the board and the support from the carrier board has a very different time zone, making it difficult to debug with them. I'll keep you informed as soon as I can test again.

mathislm, 2022-11-15:
Hello @MartyG-RealSense,

It seems the Jetpack versions required for the patch are not compatible with the Jetson Xavier AGX 64GB (as per the support, only 4.6.1/2 and 5.0.2 work). 
I suppose I'm not the first one to come up with this problem, using Realsense cameras with a Jetson board is a very classic setup. Do you have a solution for this? Maybe you already have an updated version of the Realsense patch that works ? I'm not proficient at all with these kinds of problems, while I'm sure that many of you at Intel are very good at fixing these issues.

Thank you in advance!

MartyG-RealSense, 2022-11-20:
My apologies @mathislm for the delay in responding.  I have reviewed this case again from the beginning.  

There is not an SDK update for the patch script or for supported JetPack versions available for Jetson at the time of writing this, unfortunately.  I note that installing JetPack 4.6 or 5.02 may be problematic for you due to trouble with re-flashing the board.

With JetPack versions since 4.2, the **Nvidia SDK Manager** tool can be used to perform updates.   Have you been using the SDK Manager, please?

https://developer.nvidia.com/drive/sdk-manager

https://jetsonhacks.com/2019/06/04/nvidia-sdk-manager-for-jetson-jetpack-4-2/

MartyG-RealSense, 2022-11-27:
Hi @mathislm  Do you require further assistance with this case, please?  Thanksl

closed 2022-12-04 by MartyG-RealSense