realsenseai/librealsense #11079: SR300 Depth stream "no frames received"

Leonswilly, 2022-11-09:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense SR300|
|**Serial Number**|539141001209|
|**Firmware Version**|3.26.1.0|
|**Camera Locked**|YES|
|**Product Line**|SR300|
|**Asic Serial Number**|539141001209|
|**Firmware Update Id**|539141001209|

Please provide a description of the problem
Per-fram metadata is not enabled at the OS level! Please follow the installation guide for the details

MartyG-RealSense, 2022-11-09:
Hi @Leonswilly  Are you able to launch the **RealSense Viewer** tool, please?  If you are, do you see a **Frame Metadata disabled** message pop up in the top corner of the Viewer window like in the image below?

[image]

If you do have this pop-up box then left-clicking the **Enable** button on it should enable hardware metadata support for your SR300.

If you do not see the message then unplugging the camera and then plugging it into a different USB port on the computer if you have a spare port may trigger it (it can be either a USB 3 or USB 2 port).

********

Metadata support can also be enabled by editing the Windows **registry**, though I would not recommend this unless you have knowledge of registry editing as it could break your Windows OS otherwise.

https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide#enabling-metadata-on-windows

Leonswilly, 2022-11-09:
the realsense viewer can be launched,and i did click the enable button,but still not working.is this problem relevant to the failure of installing the Intel realsense depth camera manager？

Sendtime: Wednesday, Nov 9, 2022 5:57 PM
To: "IntelRealSense/librealsense"; 
Cc: ***@***.***>; "Mention"; 

 

 

 
 
 
 
 
https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide#enabling-metadata-on-windows

Leonswilly, 2022-11-09:
i just edited the registry, still not receiving depth stream.

 
 

 
 
 
 
 
https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide#enabling-metadata-on-windows

MartyG-RealSense, 2022-11-09:
Since 2017 the Depth Camera Manager driver installation program is not required for computers with an internet connection as the camera can download and install the drivers automatically from the internet when it is plugged into the USB port.

If you are not receiving a depth stream with the SR300 after enabling it then the USB port may be providing the camera with insufficient power to start the depth stream.  This is a known issue with the SR300 model.  You can test whether your camera has a USB power issue by going to **Stereo Module > Controls** in the Viewer before enabling the stream and setting the **Laser Power** slider to zero, then enabling the depth stream.  If the depth stream is able to start with Laser Power set to zero but then disables if you increase Laser Power, the problem is likely related to insufficient power provided by the USB port.

Leonswilly, 2022-11-09:
the realsense viewer can be launched,and i did click the enable button.the rgb camera and ir mode can show pictures,but depth mode still not woking.
is this problem relevant to the failure of installing the Intel realsense depth camera manager？

Cc: ***@***.***&gt;; "Mention"; 

 
 

 
 
 
 
 
https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide#enabling-metadata-on-windows

MartyG-RealSense, 2022-11-09:
I have commented above. Thanks!

Leonswilly, 2022-11-09:
thank you for answering my questions,and sorry to bother.
it failed again...

Send time: Wednesday, Nov 9, 2022 9:01 PM

 

 
If you are not receiving a depth stream with the SR300 after enabling it then the USB port may be providing the camera with insufficient power to start the depth stream.  This is a known issue with the SR300 model.  You can test whether your camera has a USB power issue by going to Stereo Module &gt; Controls in the Viewer before enabling the stream and setting the Laser Power slider to zero, then enabling the depth stream.  If the depth stream is able to start with Laser Power set to zero but then disables if you increase Laser Power, the problem is likely related to insufficient power provided by the USB port.

MartyG-RealSense, 2022-11-09:
If you are able to enable the infrared stream then your drivers are likely installed correctly, as if there was a driver problem then only the RGB stream would be working.

I note from the title of this case that you are receiving the No Frames Received error.  This suggests to me that the depth stream is enabling and running but it is not receiving depth data.

How is the camera connected to your computer?  Are you plugging the camera directly into a USB 3 port on the computer?  The SR300 requires a USB 3 port.

Leonswilly, 2022-11-10:
it is a usb 3 port of my laptop,i tried on other two desktop pc still the same problem.dose that mean this device has something wrong.发自我的手机-------- 原始邮件 --------发件人： MartyG-RealSense ***@***.***>日期： 2022年11月9日周三 晚上9:28收件人： IntelRealSense/librealsense ***@***.***>抄送： Leonswilly ***@***.***>, Mention ***@***.***>主    题： Re: [IntelRealSense/librealsense] SR300 Depth stream "no frames received" (Issue #11079)
If you are able to enable the infrared stream then your drivers are likely installed correctly, as if there was a driver problem then only the RGB stream would be working.
I note from the title of this case that you are receiving the No Frames Received error.  This suggests to me that the depth stream is enabling and running but it is not receiving depth data.
How is the camera connected to your computer?  Are you plugging the camera directly into a USB 3 port on the computer?  The SR300 requires a USB 3 port.

MartyG-RealSense, 2022-11-10:
What happens if you go to the 'More' option near the top of the RealSense Viewer's options side-panel and select the 'Check for Updates' option from its drop-down menu to see whether the camera firmware is able to be updated?

*******************

If using 'Check for Updates' does not work for you and you wish to try the 'Depth Camera Manager' driver installation program for SR300 then it can be downloaded from the link below.

https://www.intel.com/content/www/us/en/products/sku/92329/intel-realsense-camera-sr300/downloads.html

MartyG-RealSense, 2022-11-16:
Hi @Leonswilly  Do you require further assistance with this case, please?  Thanks!

Leonswilly, 2022-11-17:
the dealer sent me another one could be arriving today. waiting to test it.发自我的手机-------- 原始邮件 --------发件人： MartyG-RealSense ***@***.***>日期： 2022年11月16日周三 傍晚5:07收件人： IntelRealSense/librealsense ***@***.***>抄送： Leonswilly ***@***.***>, Mention ***@***.***>主    题： Re: [IntelRealSense/librealsense] SR300 Depth stream "no frames received" (Issue #11079)
Hi @Leonswilly  Do you require further assistance with this case, please?  Thanks!

MartyG-RealSense, 2022-11-17:
Thanks very much for the update.  Please do let me know how the replacement works after you have tested it.

Leonswilly, 2022-11-18:
It worked.Thank's for helping.Have a good day!

Send time: Thursday, Nov 17, 2022 3:50 PM

MartyG-RealSense, 2022-11-18:
It's excellent to hear that your replacement worked.  As it works, I will close this case.  Thanks very much for the update!

closed 2022-11-18 by MartyG-RealSense