realsenseai/librealsense #11081: Memory leak in some filters

Endruu, 2022-11-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | n/a | 
| Firmware Version                   | n/a | 
| Operating System & Version | Linux (Ubuntu 20.04.5 LTS (Focal Fossa))  | 
| Kernel Version (Linux Only)    | 5.10.102.1-microsoft-standard-WSL2; 4.19.0-17-amd64 | 
| Platform                                 | PC, docker |
| SDK Version                            | 2.51.1-0~realsense0.7526 amd64 | 
| Language                            | C++17 | 
| Segment			|  | 

### Issue Description
Some filters have memory leaks in their constructor/destructors. Just creating them will result memory leaks, detected by both ASAN and Valgrind.

I've tried the following filters:
- rs2::temporal_filter - leaks
- rs2::spatial_filter - leaks
- rs2::decimation_filter - leaks
- rs2::threshold_filter - does not leak
- rs2::align - does not leak

Example program to reproduce:
```
#include <librealsense2/rs.hpp>

int main()
{
    for (int i = 0; i < 100; ++i)  // even for i = 1
    {
        rs2::temporal_filter temp_filter;          // LEAK!
        rs2::spatial_filter spat_filter;           // LEAK!
        rs2::decimation_filter dec_filter;         // LEAK!
        rs2::threshold_filter threshold_filter;    // no leak
        rs2::align align_filter(RS2_STREAM_DEPTH); // no leak
    }
}
```

Excerpts from the results:
- [rs2_leak_test_asan.log](https://github.com/IntelRealSense/librealsense/files/9974111/rs2_leak_test_asan.log)

```
Indirect leak of 64800 byte(s) in 900 object(s) allocated from:
    #0 0x55fa5841ec97 in operator new(unsigned long) (/workspaces/fg/build/asd/rs2_leak_test+0x111c97)
    #1 0x7fecf2c54ee8 in std::_Rb_tree_iterator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > std::_Rb_tree<float, std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::_Select1st<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::less<float>, std::allocator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<float const&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::piecewise_construct_t const&, std::tuple<float const&>&&, std::tuple<>&&) (/lib/x86_64-linux-gnu/librealsense2.so.2.51+0x5ceee8)

Indirect leak of 43200 byte(s) in 600 object(s) allocated from:
    #0 0x55fa5841ec97 in operator new(unsigned long) (/workspaces/fg/build/asd/rs2_leak_test+0x111c97)
    #1 0x7fecf2c54ee8 in std::_Rb_tree_iterator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > std::_Rb_tree<float, std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::_Select1st<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::less<float>, std::allocator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<float const&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::piecewise_construct_t const&, std::tuple<float const&>&&, std::tuple<>&&) (/lib/x86_64-linux-gnu/librealsense2.so.2.51+0x5ceee8)
    #2 0x7fecf2e0a6b8 in rs2_create_spatial_filter_block (/lib/x86_64-linux-gnu/librealsense2.so.2.51+0x7846b8)

Indirect leak of 20000 byte(s) in 100 object(s) allocated from:
    #0 0x55fa5841ec97 in operator new(unsigned long) (/workspaces/fg/build/asd/rs2_leak_test+0x111c97)
    #1 0x7fecf2c7bcd1 in librealsense::temporal_filter::temporal_filter() (/lib/x86_64-linux-gnu/librealsense2.so.2.51+0x5f5cd1)

Indirect leak of 20000 byte(s) in 100 object(s) allocated from:
    #0 0x55fa5841ec97 in operator new(unsigned long) (/workspaces/fg/build/asd/rs2_leak_test+0x111c97)
    #1 0x7fecf2c75630 in librealsense::spatial_filter::spatial_filter() (/lib/x86_64-linux-gnu/librealsense2.so.2.51+0x5ef630)
    #2 0x7fecf2e0a6b8 in rs2_create_spatial_filter_block (/lib/x86_64-linux-gnu/librealsense2.so.2.51+0x7846b8)
```
- [rs2_leak_test_valgrind.log](https://github.com/IntelRealSense/librealsense/files/9974109/rs2_leak_test_valgrind.log)
```
==23157== 6,001 (360 direct, 5,641 indirect) bytes in 5 blocks are definitely lost in loss record 29 of 59
==23157==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==23157==    by 0x4E2BEE8: std::_Rb_tree_iterator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > std::_Rb_tree<float, std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::_Select1st<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::less<float>, std::allocator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<float const&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::piecewise_construct_t const&, std::tuple<float const&>&&, std::tuple<>&&) (in /usr/lib/x86_64-linux-gnu/librealsense2.so.2.51.1)
==23157==    by 0x4E2C122: std::map<float, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<float>, std::allocator<std::pair<float const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >::operator[](float const&) (in /usr/lib/x86_64-linux-gnu/librealsense2.so.2.51.1)
==23157==    by 0x4E52F42: librealsense::temporal_filter::temporal_filter() (in /usr/lib/x86_64-linux-gnu/librealsense2.so.2.51.1)
==23157==   
 by 0x4FE1598: rs2_create_temporal_filter_block (in /usr/lib/x86_64-linux-gnu/librealsense2.so.2.51.1)
==23157==    by 0x10DC05: init (rs_processing.hpp:905)
==23157==    by 0x10DC05: temporal_filter (rs_processing.hpp:863)
==23157==    by 0x10DC05: main (rs2_leak_test.cpp:7)
==23157== 
==23157== 6,748 (624 direct, 6,124 indirect) bytes in 26 blocks are definitely lost in loss record 35 of 59
==23157==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==23157==    by 0x4E4698C: ??? (in /usr/lib/x86_64-linux-gnu/librealsense2.so.2.51.1)
==23157==    by 0x4E47507: librealsense::decimation_filter::decimation_filter() (in /usr/lib/x86_64-linux-gnu/librealsense2.so.2.51.1)
==23157==    by 0x4FE1478: rs2_create_decimation_filter_block (in /usr/lib/x86_64-linux-gnu/librealsense2.so.2.51.1)
==23157==    by 0x10E0A3: init (rs_processing.hpp:844)
==23157==    by 0x10E0A3: decimation_filter (rs_processing.hpp:815)
==23157==    by 0x10E0A3: main (rs2_leak_test.cpp:9)
```

MartyG-RealSense, 2022-11-09:
Hi @Endruu  At the link below there is a long and extensively detailed discussion from 2020 about Valgrind detecting memory leaks in librealsense post-processing filters, and a follow-up that the RealSense user in that case wrote.  These should be a helpful reference for you.

https://support.intelrealsense.com/hc/en-us/community/posts/360052493073-Post-processing-filters-in-C-leak-and-segmentation-fault

https://support.intelrealsense.com/hc/en-us/community/posts/360052556133-Feature-request-Valgrind-suppression-list-for-librealsense

I would recommend building your program into an executable and then using an Ubuntu system monitoring tool such as **htop** whilst your application is running to determine whether there actually is a memory leak taking place in reality.

Endruu, 2022-11-10:
I've already seen those discussions and got to the conclusion that this is a separate bug.

I've rechecked the actual memory consumption and these are actual memory leaks. Actually, we've originally detected the steady memory increase, then localized the issue to be in the temporal filter.

Regarding using unsupported kernels and building from source. I'm not that familiar with the process, so I don't know when I can look into it. Nonetheless, I don't think the leaks have anything to do with that (but this is only a gut feeling.)

MartyG-RealSense, 2022-11-10:
In a past case at https://github.com/IntelRealSense/librealsense/issues/2115#issuecomment-406869922 where a memory leak was reported in the temporal filter, it was recommended by a RealSense team member to avoid creating filters inside a streaming loop and instead only initialize them once per device.  Do you know whether the filters in your own script are within a loop, please?

Endruu, 2022-11-10:
We only use the _temporal_ and _align_ filter, however the temporal filter was indeed created in a loop. Now this is mitigated by creating it only once and reusing it. But this is only mitigation, the leak is still there, although it does not have real relevance anymore.

I feel that this issues is not that urgent, as it has a know workaround, but in this case it should be mentioned as a known issue.

Btw, going over the previous known issues once more, this seems very similar to #2860.

Endruu, 2022-11-12:
I've found the root cause. It's caused by circular referencing of shared pointers when options are setting up callbacks. They are littered through the post processing filters, but in some of them they are already fixed.

I'll create a PR fixing the rest.

MartyG-RealSense, 2022-11-12:
Thanks so much @Endruu for the update  :)  I will add an **Enhancement** label to this case to signify that it should be kept open whilst your PR is active.

Nir-Az, 2022-11-14:
PR was merged, thanks for that 👏

MartyG-RealSense, 2022-11-14:
Thanks very much @Nir-Az for the PR merge!

Nir-Az, 2022-11-14:
@Endruu May I ask for your asan + valgrind run command 
I would like to see the new output and maybe integrate it into our regression tests in the future.
Thanks!

MartyG-RealSense, 2022-11-24:
Hi @Endruu  Are you able to provide @Nir-Az with the information requested in the comment above, please?  Thanks!

Endruu, 2022-11-26:
@Nir-Az  
For valgrind, I've just built librealsense and my small test program in debug mode, then ran it with `valgrind --leak-check=full ./my_program`. For asan, it's a bit trickier and I don't have it anymore, but generally, I've used this [guideline](http://gavinchou.github.io/experience/summary/syntax/gcc-address-sanitizer/) with some adaptation to CMake.

My program looked like this:
```cpp
#include "../include/librealsense2/rs.hpp"
#include "../include/librealsense2/hpp/rs_processing.hpp"

int main()
{
    rs2::decimation_filter filter;
    rs2::disparity_transform f1;
    rs2::zero_order_invalidation f2;
    rs2::temporal_filter f3;
    rs2::spatial_filter f4;
    rs2::pointcloud f5;
    rs2::hole_filling_filter f6;

    return 0;
}
```

And the final valgrind output was:

>  valgrind --leak-check=full ./my
==3432== Memcheck, a memory error detector
==3432== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==3432== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==3432== Command: ./my
==3432== 
==3432== 
==3432== HEAP SUMMARY:
==3432==     in use at exit: 0 bytes in 0 blocks
==3432==   total heap usage: 35,029 allocs, 35,029 frees, 8,459,685 bytes allocated
==3432== 
==3432== All heap blocks were freed -- no leaks are possible
==3432== 
==3432== For lists of detected and suppressed errors, rerun with: -s
==3432== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

PR #11095 (2022-11-14): Fix memory leaks in post processing filters
Fixes #11081
closed 2022-11-27 by Nir-Az