realsenseai/librealsense #11083: How to call this  Rotation Matrix in C++

lun73, 2022-11-10:
[image]

MartyG-RealSense, 2022-11-10:
Hi @lun73  Are you aiming to retrieve the depth to color extrinsic using C++ code instead of the rs-enumerate-devices tool, please?  Thanks!

lun73, 2022-11-10:
yes! I using C++ code

MartyG-RealSense, 2022-11-10:
At the RealSense documentation link below, under the 'librealsense2' title there is a C++ code example for getting the depth to color extrinsic with the following instruction:

`rs2_extrinsics e = depth_stream.get_extrinsics_to(color_stream);`

https://dev.intelrealsense.com/docs/api-how-to#get-and-apply-depth-to-color-extrinsics

lun73, 2022-11-10:
`auto stream_depth = pipe.get_active_profile().get_stream(RS2_STREAM_DEPTH);`
`auto stream_color = pipe.get_active_profile().get_stream(RS2_STREAM_COLOR);`
`rs2_extrinsics extrinsics = stream_color.get_extrinsics_to(stream_depth);`
`rs2_extrinsics extrinsics_1 = stream_depth.get_extrinsics_to(stream_color);`
`cv::Matx33f main_DtoC = { extrinsics.rotation[0],extrinsics.rotation[1],extrinsics.rotation[2],
								extrinsics.rotation[3],extrinsics.rotation[4],extrinsics.rotation[5], 
								extrinsics.rotation[6],extrinsics.rotation[7],extrinsics.rotation[8], };`
The result is depth to color only when I use the third line.
Shouldn't I use the third line?

MartyG-RealSense, 2022-11-10:
A number of references state that **depth_stream.get_extrinsics_to(color_stream)** is the depth to color extrinsics instruction.

The official RealSense C# version of the extrinsics example also places **color** in the brackets.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#get-and-apply-depth-to-color-extrinsics

[image]

MartyG-RealSense, 2022-11-16:
Hi @lun73  Do you require further assistance with this case, please?  Thanks!

lun73, 2022-11-17:
`auto stream_depth = pipe.get_active_profile().get_stream(RS2_STREAM_DEPTH);`
`auto stream_color = pipe.get_active_profile().get_stream(RS2_STREAM_COLOR);`
`rs2_extrinsics extrinsics = stream_color.get_extrinsics_to(stream_depth);`
`rs2_extrinsics extrinsics_1 = stream_depth.get_extrinsics_to(stream_color);`
`cv::Matx33f main_DtoC = { extrinsics.rotation[0],extrinsics.rotation[1],extrinsics.rotation[2],
                                            extrinsics.rotation[3],extrinsics.rotation[4],extrinsics.rotation[5],
					    extrinsics.rotation[6],extrinsics.rotation[7],extrinsics.rotation[8] };`
`cv::Matx31f main_Translation = { extrinsics_1.translation[0],extrinsics_1.translation[1] ,extrinsics_1.translation[2] };`
`cout << "DtoC_rotation" << main_DtoC << endl;`
`cout << "DtoC_translation" << main_Translation << endl;`
My Code
[image]

rs-enumerate-devices.exe
[image]

Why do I have to use COLOR TO DEPTH to run DEPTH TO COLOR?

MartyG-RealSense, 2022-11-17:
The advice provided by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/5326 gives me the impression that the COLOR TO DEPTH extrinsic may be the appropriate one to use if depth to color alignment is **not** being used.

lun73, 2022-11-18:
So it is correct to say that this usage?

MartyG-RealSense, 2022-11-18:
For a non-aligned image, my judgement would be Yes.

lun73, 2022-11-18:
> 對於未對齊的圖像，我的判斷是肯定的。

Thank you for your reply.

MartyG-RealSense, 2022-12-02:
Case closed due t no further comments received.

closed 2022-12-02 by MartyG-RealSense