realsenseai/librealsense #11087: One of my D435 camera gives bad depth quality

alperenyavascan1, 2022-11-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.51.1.0                          | 

Hello, 

I have three D435 cameras, but one of them gives me bad depth result, and all of them has same calibration data and same configuration file. Can you give me advice how I can improve this camera depth quality, what do I need to do.

As seen in the pictures below, the upper picture gives a noisy image, while the lower picture gives a very good result.
Is the camera is broken or something else?

[image]
[image]
[image]

MartyG-RealSense, 2022-11-10:
Hi @alperenyavascan1  I observe from the scale at the right edge of the image that the good images appear to be limiting the observable depth distance to a maximum of 4 meters, whilst on the 'bad' images the camera is able to sense to a further distance.

Please open the **Post Processing** list of filters in the RealSense Viewer (under the Stereo Module category) and check whether the **Threshold** filter is disabled (a red icon beside it).  If it does have the red 'disabled' icon beside the Threshold filter, please left-click on it to turn it blue (enabled) and see whether the images improve.

alperenyavascan1, 2022-11-11:
Thank you for answer @MartyG-RealSense  but I try to open threshold filter and the other filters but nothing can solve my problem as you can see in the below images, so do you have any other suggestions?
[image]
[image]
[image]
[image]

MartyG-RealSense, 2022-11-11:
The majority of the break-up in the images seems to be at the nearest area to the camera lenses (at the bottom of the screen).

Is the affected camera positioned at the same distance from the observed scene as the camera with the good images?  If the affected camera was further forwards than the other one then that kind of break-up could be caused by close-range surfaces being nearer to the camera lenses than the D435's minimum depth sensing distance of around 0.1 m / 10 cm.

If the affected camera is at the same distance from the scene as the other camera, please try resetting the camera's calibration to its factory-new default values in the RealSense Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

alperenyavascan1, 2022-11-11:
Yes two cameras are in the same place and same distance, both of the cameras are 60-70 cm above the ground. 
I did resetting the camera's calibration to factory values in realsense-viewer but still the depth quality is not change.

MartyG-RealSense, 2022-11-11:
Are the cameras mounted together vertically, one on top of another?  If they are, is the camera with the bad images the one that is mounted lower?

MartyG-RealSense, 2022-11-11:
Would it be possible to try the affected camera with another RealSense application such as **rs-capture** or **rs-multicam** to test whether the depth image problem only affects the RealSense Viewer or occurs in other tools as well?

MartyG-RealSense, 2022-11-16:
Hi @alperenyavascan1  Do you require further assistance with this case, please?  Thanks!

alperenyavascan1, 2022-11-17:
Hi @MartyG-RealSense, still my issue is continue, I increased the laser power, but instead of getting smaller, the holes got bigger and the noise increased. I also tried rs-multicam but still my issue is not solved.

MartyG-RealSense, 2022-11-17:
Have you tried swapping the USB cable on the affected camera with a cable from a good camera to eliminate the possibility of a damaged USB cable, please?

alperenyavascan1, 2022-11-18:
Yes, I already try this but nothing change.

MartyG-RealSense, 2022-11-18:
If increasing Laser Power increases the black holes, does reducing the Laser Power value make the holes smaller and fewer?

Could you also check please whether there are dots on the RGB image.  There should not be but in rare cases the infrared dot pattern can 'leak' out into the RGB image.

alperenyavascan1, 2022-11-18:
Yes If I reducing the Laser Power the holes become smaller and if I turn off the laser I don't see any holes but overall noise increase a lot and I want to use laser of the camera.

I check the RGB image and I didn't see any leak.

MartyG-RealSense, 2022-11-18:
You could set Laser Power to '1' to keep the projector active but the dot pattern invisible (the laser shuts off if Laser Power is '0').  There may be a loss of depth detail from minimizing Laser Power, but this can be compensated for if the scene that the camera is observing is well illuminated, as 400 Series cameras can alternatively use a scene's ambient light to aid depth analysis of surfaces instead of analyzing the dot pattern that is cast onto surfaces.

alperenyavascan1, 2022-11-22:
So, If I understand correctly, this camera is not working properly and the problem is not software,it is laser, am I right? I try many configuration but none of them give me a proper image.

MartyG-RealSense, 2022-11-22:
If you have three cameras with the same configuration and two of them provide a good image in the Viewer but the third camera doesn't and the depth image worsens instead of improves when Laser Power is increased, that would suggest a problem that cannot be corrected with software or drivers.

If the affected camera was purchased during the past year then you could consider contacting the retailer that you purchased the camera from to enquire about a return of the camera.

MartyG-RealSense, 2022-11-27:
Hi @alperenyavascan1  Do you require further assistance with this case, please?  Thanksl

MartyG-RealSense, 2022-12-04:
Case closed due to no further comments received.

closed 2022-12-04 by MartyG-RealSense

puyiwen, 2023-01-16:
> If you have three cameras with the same configuration and two of them provide a good image in the Viewer but the third camera doesn't and the depth image worsens instead of improves when Laser Power is increased, that would suggest a problem that cannot be corrected with software or drivers.

Hi, I use D435i and find that at sometimes RGB images have some black and white point noise, and the corresponding depth map will have many polka dots, which will make the depth map very uneven. Why is that? How should I set my RGB camera parameters?I am using the default.json now. Can you help me? Thank you very much!!

puyiwen, 2023-01-16:
> 

[image]
The image above is an example image

MartyG-RealSense, 2023-01-16:
Hi @puyiwen  There are rare occasions when the infrared dot pattern will leak through to the RGB image when it should not do so.  Excessive dots on the depth image could also be a symptom of a noise phenonemon called **laser speckle** that results from the camera's infrared dot pattern projector being based on a laser.  

In both cases, dots can be reduced by reducing the value of the **Laser Power** setting to make the dots that are cast onto the observed scene by the IR projector less visible to the camera.

Reducing laser power can though result in reduced quality of the depth image, such as more gaps / holes.  Increasing the external light illumination in the scene may help to compensate for this.  This is because 400 Series cameras can use the ambient light in the scene to analyze surfaces for depth information instead of analyzing the visible dot pattern.

puyiwen, 2023-01-16:
Thank you for your quick reply! As you said, I should brighten the light source in the scene to avoid this. Can I increase the **brightness**,which in **RGB Controls** to mitigate this?

puyiwen, 2023-01-16:
> Hi @puyiwen There are rare occasions when the infrared dot pattern will leak through to the RGB image when it should not do so. Excessive dots on the depth image could also be a symptom of a noise phenonemon called **laser speckle** that results from the camera's infrared dot pattern projector being based on a laser. Using an external **LED** projector nstead of the camera's built-in laser projector can help to avoid speckle.
> 
> In both cases, dots can be reduced by reducing the value of the **Laser Power** setting to make the dots that are cast onto the observed scene by the IR projector less visible to the camera.
> 
> Reducing laser power can though result in reduced quality of the depth image, such as more gaps / holes. Increasing the external light illumination in the scene may help to compensate for this. This is because 400 Series cameras can use the ambient light in the scene to analyze surfaces for depth information instead of analyzing the visible dot pattern.

[image]

MartyG-RealSense, 2023-01-16:
The **Brightness** setting will increase the brightness of the RGB image but will not affect the level of external illumination in the room, so would not improve calibration results.

If your scene is bright then you could try disabling the IR emitter to remove the dot pattern completely and see whether the location's illumination is sufficient to still provide a good quality depth image.

puyiwen, 2023-01-17:
> 

Sorry to bother you again, I set the **Emitter Enabled** off, and I find the depth map like that.
[image]
[image]
Are they seems true?

MartyG-RealSense, 2023-01-17:
The way in which the depth shading changes progress smoothly from blue in the foreground towards red in the furthest distance seems as though it is a reliable depth image.

puyiwen, 2023-01-17:
Thank you very much! I take some outdoor scenes with D435i, and I find that the part of the building could not be identified. Is it due to the distance of the building or is it due to reflections?

 

浦一雯
***@***.***

puyiwen, 2023-01-17:
> The way in which the depth shading changes progress smoothly from blue in the foreground towards red in the furthest distance seems as though it is a reliable depth image.

[image]
The photo is here.

MartyG-RealSense, 2023-01-17:
Reflections of glass windows can be difficult for the camera to read, but it should be able to detect the concrete wall parts of the building.  Please check in the **Post-Processing** list of filters if a filter called **Threshold Filter** is enabled.  This filter limits the observable depth range of the camera to 4 meters by default.  Disabling the filter will enable the camera to render depth detail as far as it is able to see.

puyiwen, 2023-01-17:
Thanks, I will take your advice. I took some pictures in a slightly darker room in order to avoid the effect of fluorescent lights, and I found that the depth map still have noise.I also adjusted the exposure and use IR, depth map remains unchanged. How do I smooth out the depth map?
 

浦一雯
***@***.***

MartyG-RealSense, 2023-01-17:
If the depth values are fluctuating instead of remaining stable then reducing the **filter smooth alpha** setting of the post-processing **Temporal Filter** **from its default of '0.4' to '0.1' can stabilize the depth readings.   This method is best used for situations where the camera is observing a non-moving object.

puyiwen, 2023-01-18:
OK, I guess the noise of depth map because the distance of the object. I find that when the object is close to the camera, the depth map is nomal, is my guess correct? My indoor scene takes up to 10m, is it already exceeding the optimal distance of the depth map? I wonder what is the range of the best shooting distance for D435i depth camera? Thank you again!!
 

浦一雯
***@***.***

MartyG-RealSense, 2023-01-18:
The D435i camera model can depth-sense up to 10 meters.  However, due to a phenomenon called **RMS Error**, error starts at around zero at the camera lenses and increases linearly as the distance from the camera of an observed object / surface increases.  On the D435i, this drift in accuracy starts to become noticable at around 3 meters distance and greater.

The D455 model has 2x the accuracy over distance of D435i, so at 6 meters the D455 has the same accuracy that the D435i has at only 3 meters.

In long range indoor scenes it may help to use an external dot pattern projector if possible, as it will have a greater range than the camera's built-in projector.  This subject is discussed at https://github.com/IntelRealSense/librealsense/issues/8258

Intel also have a white-paper guide about use of projectors with 400 Series cameras at the link below.

https://dev.intelrealsense.com/docs/projectors