realsenseai/librealsense #11090: How to get CSV snapshots

singchee, 2022-11-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 2.51.1.4348 | 
| Operating System & Version |   Windows 11 | 

Hi Everyone

I'm really new to using the D415, and have no programming experience so am struggling.

What I'm trying to do
1. Take still images of people's face and generate a depth map
2. Have the depth map represented as a CSV file (or equivalent) that  can be used for analysis of the facial contours

From reading the posts, it seems this I can do this by
1. Taking a "snapshot" in the 2D mode of the Realsense viewer - this exports to a .raw file
2. Use a script to get the depth data from each pixel - as described here: https://github.com/IntelRealSense/librealsense/issues/2200
however, what I need help understanding is
- What programming language is needed to extract the depth data for each pixel (e.g. in the link above, what language is used)? 
- How is the depth data represented? 
- how can i get it into a CSV format?

Thank you very much!

Sing

MartyG-RealSense, 2022-11-11:
Hi @singchee  If you are using the RealSense Viewer's 'Snapshot' feature to capture the depth stream then you should not need to convert a raw file to .csv as the Snapshot already exports a .csv file as part of the three file types that it creates (.png, .raw and .csv).

The .csv file's icon's appearance will depend on which program your computer is currently set to open .csv files, though you can tell which of the three exported files is a .csv from "csv" on the end of its filename.

[image]

If you wish to explore how to export a .csv then a modification of the RealSense SDK's C++ **rs-capture** example program at the link below that was created by a RealSense team member that exports data to .csv may be a helpful programming reference.

https://github.com/dorodnic/librealsense/tree/text_sample/examples/capture

https://github.com/dorodnic/librealsense/blob/text_sample/examples/capture/rs-capture.cpp

In regard to how depth data is represented, it is a set of coordinates with a numeric value in meters that represents the distance of a particular coordinate on the depth image from the camera.   For example, 50 cm from the camera would be '0.5' meters.

singchee, 2022-11-11:
Thanks Marty, yes agree the .CSV with the export function is very limited.

Is there a guide on how to extract depth data from the raw file? The links provided above are extracting data from the feed.

Thanks again!

MartyG-RealSense, 2022-11-11:
There is a recent discussion at the link below regarding reading depth data from a .raw file.

https://support.intelrealsense.com/hc/en-us/community/posts/11024587584915-How-to-get-the-depth-information-from-raw-file-python-

In regard to RealSense .ply point cloud files, you can import them into other software tools such as **MeshLab** or into specialized pointcloud libraries such as PCL in order to do further work with them , but it is not practical to load them back into the RealSense SDK directly.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

If you wanted to capture a single-frame snapshot that preserves the depth data perfectly in a format that can be directly read by an SDK script as though it were live camera data then you could use a script with the **save_single_frameset()** instruction to save a single depth frame to a **bag** format file.  The link below has a complete Python script that demonstrates this function.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/single_frameset.py

closed 2022-11-24 by MartyG-RealSense