realsenseai/librealsense #11091: How to get imformation of realsence camera

kankanzheli, 2022-11-11:
I want to ask, where can I learn the specific information of the realpresence camera? It includes weight and depth to obtain distance information.

And most importantly, what is the depth information acquisition method of Realsence camera? I found that the current camera uses binocular match to obtain depth information. Is infrared ranging used?

The cameras I use include D415, D425, D435 and D425i

kankanzheli, 2022-11-11:
And D435i. Not D425i

MartyG-RealSense, 2022-11-11:
Hi @kankanzheli  RealSense 400 Series cameras are **stereo depth** technology, which means that they construct the depth frame from left and right infrared images.  These are raw infrared images in the camera hardware and not the same as the 'Infrared' and 'Infrared 2' streams.

Intel have an introductory **Beginner's Guide to Depth** article at the link below that provides information about the principles of stereo depth.

https://www.intelrealsense.com/beginners-guide-to-depth/

Comparisons of RealSense camera models, including their depth sensing ranges, can be found at the links below:

https://www.intelrealsense.com/compare-depth-cameras/

https://www.intelrealsense.com/which-device-is-right-for-you/

Page 66 of the current edition of the data sheet document for the 400 Series cameras lists the weight for each model.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

The information is quoted in the tables below.

[image]

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.

closed 2026-10-23 by MartyG-RealSense