realsenseai/librealsense #11094: Cant get the viewer to run

yannickErp, 2022-11-11:
After running `go run cmd/viewer/main.go`
I get this:
`# github.com/go-gl/glfw/v3.1/glfw
In file included from ../../go/pkg/mod/github.com/go-gl/glfw@v0.0.0-20221017161538-93cebf72946b/v3.1/glfw/cocoainit_darwin.go:6:
In file included from ./glfw/src/cocoa_init.m:27:
./glfw/src/internal.h:56:3: error: "No supported client library selected"
 #error "No supported client library selected"
  ^
1 error generated.`

I am on MacOS 12.6

MartyG-RealSense, 2022-11-11:
Hi @yannickErp  Which method did you use to install the RealSense SDK on MacOS 12.6 (Monterey) please?  I would recommend using one of the methods below for a Monterey installation.  Both are compatible with Apple Silicon M1 models of Mac if you have one of those Macs.

1.  A detailed RealSense installation guide for Monterey and M1.
 
https://lightbuzz.com/realsense-macos/

2.  The **brew install librealsense** installation method.
https://formulae.brew.sh/formula/librealsense

*********

Alternatively, as the error message references **GLFW**, you could try installing GLFW (which the RealSense SDK uses for graphics) with the following command:

**brew install homebrew/versions/glfw3**

yannickErp, 2022-11-11:
Hi @MartyG-RealSense I installed the RealSense with the homebrew command `brew install librealsense`, and GLFW with `brew install glfw3`. 
The result is still the same error sadly.

MartyG-RealSense, 2022-11-11:
https://github.com/IntelRealSense/librealsense/issues/2811#issuecomment-443909901 suggests a method for launching realsense-viewer after having installed with `brew install librealsense`

MartyG-RealSense, 2022-11-18:
Hi @yannickErp  Do you require further assistance with this case, please?  Thanks!

yannickErp, 2022-11-20:
@MartyG-RealSense thank you for your help, reinstalling librealsense with `brew reinstall librealsense` worked for me!

MartyG-RealSense, 2022-11-20:
That's great to hear, @yannickErp - thanks very much for the update!  As you achieved a solution, I will close this case.  Thanks again!

closed 2022-11-20 by MartyG-RealSense