realsenseai/librealsense #11096: unable to locate packages

dhruvinvadgama, 2022-11-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |Linux (Ubuntu 22.04) | 

### Issue Description
I am trying to install the Realsense SDK from source on Ubuntu 22.04.

When I try to install the libraries and the developer & debug packages, it says, "unable to locate packages". 
- sudo apt-get install librealsense2-utils
- sudo apt-get install librealsense2-dev
- sudo apt-get install librealsense2-dbg

I have installed this SDK multiple times in the last few weeks but I didn't see this problem before. 

[image]

MartyG-RealSense, 2022-11-13:
Hi @dhruvinvadgama   Have you previously attempted installation of the librealsense SDK on Ubuntu 22.04, please?  The SDK does not yet officially support 22.04 at the time of writing this.  A pre-built test package for it is available at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106

dhruvinvadgama, 2022-11-13:
Hello,
Yes I have previously used the SDK on Ubuntu 22.04 and It worked fine. I have done this multiple times on 6-7 different computers with same environment. Only yesterday it just doesn't seem to find the packages anymore..

MartyG-RealSense, 2022-11-13:
The message **librealsense2-dkms is already the newest version** makes me think that it is already installed on the computer.

You could try removing all RealSense related packages from your computer with the command below and then trying installation again.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

dhruvinvadgama, 2022-11-13:
Okay. I will try that.
Also, as you can see in the photo, the librealsense2-dkms could be installed, but librealsense2-utils, librealsense-dev and librealsense-dbg are not found.

MartyG-RealSense, 2022-11-13:
librealsense2-dkms contains the core of the RealSense SDK.  librealsense2-utils installs RealSense example programs and tools, such as realsense-viewer.  

librealsense2-dev and librealsense2-dbg are optional installs for developer activities such as debugging and building an application with g++ or your own choice of IDE.

What kernel version are you using, with 22.04 please?

dhruvinvadgama, 2022-11-13:
I am using 5.15.0-52-generic

MartyG-RealSense, 2022-11-13:
Up until now, RealSense dkms packages have officially supported a maximum kernel version of **5.4** rather than 5.15.  The RealSense developer team made some changes in the past week to support kernels 5.13 and 5.15 with dkms - as described at the PR page https://github.com/IntelRealSense/librealsense/pull/11076/files

I would recommend trying the installation again tomorrow (Monday) to see whether the problem with the packages is related to a weekend issue or whether they remain unavailable to you.

dhruvinvadgama, 2022-11-13:
Okay, I will try installing again tomorrow to find out the issue.
 
Yes, I am using the official distribution_linux,md instructions to install.
Also, I am using v2.50.0

MartyG-RealSense, 2022-11-16:
Have you had an opportunity to perform the installation, please?  Thanks!

tanguy-magne, 2022-11-23:
I am facing the same issue with kernel version 5.15.0-53-generic.
Any updates?
Thanks!

hjwbucea, 2022-11-23:
> Okay, I will try installing again tomorrow to find out the issue.
> 
> Yes, I am using the official distribution_linux,md instructions to install. Also, I am using v2.50.0

Hello, have you resolved this issue now, I am facing the same problem, thanks

dhruvinvadgama, 2022-11-26:
Hello,

Not really, but I have found a way around. 
Here is the official Linux installation.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

I have tried to build librealsense with the RSUSB backend, which doesn't depend on the linux kernels. I found this information in an issue which was posted here.
https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1107521661

I still could not install the following packages.
    librealsense2-utils
    librealsense2-dev
    librealsense2-dbg

closed 2022-11-26 by dhruvinvadgama

MartyG-RealSense, 2022-11-26:
Hi @dhruvinvadgama  If librealsense is built from source code with the RSUSB backend using CMake then including the build flags **-DBUILD_EXAMPLES** and **DBUILD-GRAPHICAL-EXAMPLES** in the CMake build instruction will build the tools and examples in addition to the librealsense SDK.  This has the same effect as installing them from the **librealsense2-utils** package.

librealsense will be built in Debug mode by default when building from source code.   It can be built in 'Release' mode with optimizations included by using the build flag **-DCMAKE_BUILD_TYPE=release**

These instructions are demonstrated in the instruction guide at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692 for building with the RSUSB backend.

dhruvinvadgama, 2022-11-26:
I searched for the packages with 'sudo find' and I could not locate them.  I understand now that the graphics support, tools and examples are installed, not inside the above listed packages but separately. Correct?

Also, the resizing of the frames is not possible. I cannot change the size from (640 x 480) to other sizes like (1280 x 720) etc. Is this problem also linked to the package installation or is it completely different?

MartyG-RealSense, 2022-11-26:
Package installation and building from source code are two different methods of achieving the same goal - installing the librealsense SDK.  One or the other should be chosen but not both performed on the same computer, otherwise you could end up with two installations of librealsense that conflict with each other.

So if your goal, as you mentioned at the start of this case, is to install from source code then you should completely ignore the package installation instructions at the link below and instead use CMake to build from source instead. You do not need to use the packages at all to achieve a successful source build.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

*****

In regard to changing the frame size, do you mean changing the resolution of a stream?  If so then you cannot do this while the stream is enabled.  You need to stop it, change the resolution and enable the stream again.

In the RealSense Viewer tool this is simple to do with the side-panel controls.  In a script it would be done by using a Stop() instruction to stop the pipeline, and then use cfg instructions after the Stop() line to define a new stream resolution and Start() the pipeline again.

dhruvinvadgama, 2022-11-27:
Okay. Thank you. I think this helps.