realsenseai/librealsense #11097: Connecting Realsense L515 in Network

TheNemo05, 2022-11-14:
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version | Ubuntu 20, Debian 11 | 
| Kernel Version (Linux Only)    |  for ubuntu (Linux 5.14.0-1054-oem x86_64), for debian (Linux 5.15.76-v8+ aarch64)| 
| Platform                                 | PC & Raspberry Pi |
| SDK Version                            | v2.51.1 | 
| Language                            | Python | 

### Issue Description
As mentioned in system parameters 2 systems are being used, Ubuntu installed in PC & Debian based system used in Raspberry pI Model 4. Both the systems has Realsense SDK viewer version 2.51.1. Realsense L515 lidar is connected to Raspberry Pi using USB type C connection.  Raspberry pi is connected to PC using SSh command. 

[image]
The IP 10.42.0.176 is Raspberry Pi's IP.
As per the process shown in https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras the Realsense SDK should display the camera connected in Network using the Add network device. 
[image]
[image]
After inserting the IP of Raspberry pi, an error is displayed stating '[getStreams] error: Connection to server failed: Connection refused - -107' 
Is there anything missing, Thanks in advance for support.

TheNemo05, 2022-11-14:
@MartyG-RealSense can u please look into this.

MartyG-RealSense, 2022-11-14:
Hi @TheNemo05  Can you confirm please whether you have built the librealsense SDK on the Pi 4 from source code with CMake and included **-DBUILD_NETWORK_DEVICE=ON** in the CMake build instruction so that the SDK is built with networking support and installs the **rs-server** networking tool?  Doing this is necessary in order to be able to make a network connection from the RealSense Viewer to the camera.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#32-building-from-source

TheNemo05, 2022-11-14:
i ve included the -DBUILD_NETWORK_DEVICE=ON in CMake file, i ve not found any document to install rs-server. can u help me with it please.

MartyG-RealSense, 2022-11-14:
rs-server should be compiled automatically when the SDK is built if ** -DBUILD_NETWORK_DEVICE=ON** is included in the CMake build instruction on the Pi.  For example:

`cmake ../ -DCMAKE_BUILD_TYPE=RELEASE -DBUILD_NETWORK_DEVICE=ON`

However, a RealSense team member advises at https://github.com/IntelRealSense/realsense-ros/issues/1804#issuecomment-883900014 that the L515 camera model is not supported by rs-server.  As the L515 model is now End of Life (retired), support would not be added to rs-server at a future date.

MartyG-RealSense, 2022-11-14:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

TheNemo05, 2022-11-14:
After cloning the repository & running this command 'sudo python setup.py' from librealsense directory, an error message is displayed 'python: can't open file 'setup.py': [Errno 2] No such file or directory'

MartyG-RealSense, 2022-11-16:
The EtherSense project is also hosted on its developer's own GitHub.  Do you experience the same problem if you clone that repo, please?

https://github.com/krejov100/EtherSense

MartyG-RealSense, 2022-11-27:
Hi @TheNemo05  Do you require further assistance with this case, please?  Thanksl

closed 2022-12-04 by MartyG-RealSense