realsenseai/librealsense #11099: Does sdk support windows 11 system?

2513297244, 2022-11-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {win11}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<50>.<0> }                          | 
| Language                            |  {C}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Does sdk support windows 11 system? The problem of camera opening failure frequently occurs on Windows 11 system

TheNemo05, 2022-11-14:
yes, use USB 3 port for accessing camera.

MartyG-RealSense, 2022-11-14:
Hi @2513297244  Yes, the RealSense SDK works with Windows 11, with a very small number of cases where problems have been experienced.  Can you provide more information please about the circumstances in which the camera opening failures occur?

2513297244, 2022-11-14:
> Hi @2513297244 Yes, the RealSense SDK works with Windows 11, with a very small number of cases where problems have been experienced. Can you provide more information please about the circumstances in which the camera opening failures occur?

I use wait when the camera is turned on_ for_ The frames function will throw an exception to remind me that no frame has been received within the specified time

2513297244, 2022-11-14:
> Hi @2513297244 Yes, the RealSense SDK works with Windows 11, with a very small number of cases where problems have been experienced. Can you provide more information please about the circumstances in which the camera opening failures occur?

After the camera is turned on, I use the wait_for_frames() function to throw an exception prompt, saying that I have not received a frame within the specified time

MartyG-RealSense, 2022-11-14:
Does the camera perform normally in the RealSense Viewer tool?

Would it be possible to paste your script into a comment below, please?

closed 2022-11-27 by MartyG-RealSense