realsenseai/librealsense #11101: How can I get point cloud only with color and depth map？

qq996235826, 2022-11-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i } | 
| Operating System & Version |   {Win 11}  | 
| Platform                                 | PC |
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

I used this method before to get point cloud          ->  pyrealsense2.pointcloud.calculate(depth)
but now, I need to get point cloud by depth map, can this be done, is there any method to do it？
or can I save the frame as a file? so that I can continue to use the method to get a point cloud.
thanks!

qq996235826, 2022-11-14:
> Hi @qq996235826 An alternative method to pc.calculate for obtaining a point cloud is to use depth-to-color alignment first and then use the instruction **rs2_deproject_pixel_to_point** secondly, as demonstrated in the Python script at #10438
> 
> Using pc.calculate and then retrieving the XYZ coordinates from its **vertices** is more accurate than obtaining XYZ via rs2_deproject_pixel_to_point though as it is the depth-color alignment process that can introduce inaccuracy.
> 
> It is possible to export a color pointcloud to a **.ply** pointcloud file in Python, though the only method for doing so that is confirmed to work is the script at [#6194 (comment)](https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293) that uses pc.calculate.
> 
> There is another method of exporting to a ply file called **save_to_ply** that gives greater control over configuration of the file (such as color / no color and binary / ASCII). There is a Python export script at [#7747 (comment)](https://github.com/IntelRealSense/librealsense/issues/7747#issuecomment-725346152) (again, it uses pc.calculate) though it did not export color successfully when last tested in 2020. You are welcome to test it to see whether it works for you now.

Thank you for your answer. My problem has been solved

closed 2022-11-14 by qq996235826

MartyG-RealSense, 2022-11-14:
I'm pleased that I could help.  Thanks very much for the update!