realsenseai/librealsense #11104: Adding Lens in front of RGB Camera | Realsense D400

farhanoktvn, 2022-11-14:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.51.1 | 
| Language                            |  Python | 

### Issue Description
Hello, we're doing a project where we need to have the camera close to the object. Since the camera can only focus from 20cm to infinity, we are planning to add a lens in front of the camera to improve the near-sight focus. I understand the camera will need to be recalibrated. My question is does the calibration provided in the SDK sufficient to calibrate the camera with the added lens or do we need to derive our own camera extrinsic and intrinsic values?

Thank you for your help.

MartyG-RealSense, 2022-11-14:
Hi @farhanoktvn  The **RealSense D405** camera model may suit your project.  It is designed for high quality, high accuracy images at very close range.  Its default ideal depth sensing range is 7 cm to 50 cm, though adjusting a setting called **Disparity Shift** can enable it to sense at 4 cm range, as described at https://github.com/IntelRealSense/librealsense/issues/10963#issuecomment-1274941411

There was a past case at https://github.com/IntelRealSense/librealsense/issues/7631 where a RealSense user considered whether they could alter a D435 model to improve close-range image quality.  They were advised by a RealSense team member that the optics of a non-D405 camera cannot be retrofitted to change focus.

farhanoktvn, 2022-11-15:
Hi @MartyG-RealSense,

Thank you for your answer. I do not need further help with this issue.

closed 2022-11-15 by farhanoktvn

MartyG-RealSense, 2022-11-15:
You are very welcome, @farhanoktvn - thanks for the update!