realsenseai/librealsense #11106: librealsense won't build on 5.11.0 / 20.04

dheera, 2022-11-15:
```
Removing old librealsense2-dkms-1.3.19 DKMS files...

------------------------------
Deleting module version: 1.3.19
completely from the DKMS tree.
------------------------------
Done.
Loading new librealsense2-dkms-1.3.19 DKMS files...
Building for 5.11.0-27-generic 5.15.0-52-generic
Building initial module for 5.11.0-27-generic
Error! Bad return status for module build on kernel: 5.11.0-27-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.19/build/make.log for more information.
dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

make.log contains:
```
DKMS make.log for librealsense2-dkms-1.3.19 for kernel 5.11.0-27-generic (x86_64)
Tue 15 Nov 2022 01:26:19 AM GMT
make: Entering directory '/usr/src/linux-headers-5.11.0-27-generic'
scripts/Makefile.build:44: /var/lib/dkms/librealsense2-dkms/1.3.19/build/5.11.0/drivers/media/usb/uvc/Makefile: No such file or directory
make[1]: *** No rule to make target '/var/lib/dkms/librealsense2-dkms/1.3.19/build/5.11.0/drivers/media/usb/uvc/Makefile'.  Stop.
make: *** [Makefile:1848: /var/lib/dkms/librealsense2-dkms/1.3.19/build/5.11.0/drivers/media/usb/uvc] Error 2
make: Leaving directory '/usr/src/linux-headers-5.11.0-27-generic'
```

The file `/var/lib/dkms/librealsense2-dkms/1.3.19/build/5.11.0/drivers/media/usb/uvc/Makefile` does not exist.
The file `/var/lib/dkms/librealsense2-dkms/1.3.19/build/5.15.0/drivers/media/usb/uvc/Makefile` DOES exist.

MartyG-RealSense, 2022-11-15:
Hi @dheera  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/5366 who experienced the **Bad return status for module build on kernel** error message found that the solution in their particular case was to reinstall everything on a clean Ubuntu installation.  

You can remove all RealSense related DKMS packages from your computer with the command below:

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

MartyG-RealSense, 2022-11-18:
Hi @EthanTwo2022  Kernel 5.11 is supported for **source code** builds of the **2.51.1** librealsense SDK.

MartyG-RealSense, 2022-11-24:
Hi @dheera and @EthanTwo2022  Do you require further assistance with this case, please?  Thanks!

MartyG-RealSense, 2022-12-02:
Case closed due to no further comments received.

closed 2022-12-02 by MartyG-RealSense

YXGuan, 2022-12-18:
Hi Marty, 

I have received the same error. Tried removing all Realsense libraries but the same error persists.

> Case closed due to no further comments received.

MartyG-RealSense, 2022-12-18:
Hi @YXGuan  Which computer / computing device are you installing librealsense from packages on, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc).  Thanks!

YXGuan, 2022-12-18:
> Hi @YXGuan Which computer / computing device are you installing librealsense from packages on, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc). Thanks!

Thanks for your prompt response

Laptop Lenovo Carbon X1

Ubuntu 20.04

YXGuan, 2022-12-18:
from /var/crash/librealsense2-dkms.0.crash:
`ProblemType: Package
DKMSBuildLog:
 DKMS make.log for librealsense2-dkms-1.3.19 for kernel 5.14.0-1054-oem (x86_64)
 Sun 11 Dec 2022 03:20:14 PM EST
 make: Entering directory '/usr/src/linux-headers-5.14.0-1054-oem'
 scripts/Makefile.build:44: /var/lib/dkms/librealsense2-dkms/1.3.19/build/5.14.0/drivers/media/usb/uvc/Makefile: No such file or directory
 make[1]: *** No rule to make target '/var/lib/dkms/librealsense2-dkms/1.3.19/build/5.14.0/drivers/media/usb/uvc/Makefile'.  Stop.
 make: *** [Makefile:1870: /var/lib/dkms/librealsense2-dkms/1.3.19/build/5.14.0/drivers/media/usb/uvc] Error 2
 make: Leaving directory '/usr/src/linux-headers-5.14.0-1054-oem'
DKMSKernelVersion: 5.14.0-1054-oem
Date: Sun Dec 11 15:20:15 2022
Package: librealsense2-dkms 1.3.19-0ubuntu1
PackageVersion: 1.3.19-0ubuntu1
SourcePackage: librealsense2
Title: librealsense2-dkms 1.3.19-0ubuntu1: librealsense2-dkms kernel module failed to build`

MartyG-RealSense, 2022-12-18:
The librealsense SDK does not support kernel 5.14 unfortunately.  It supports the kernels either side of it, 5.13 and 5.15.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages

If you are unable to change your kernel version then it is possible to bypass the kernel by building the SDK from source code with CMake using the **RSUSB backend** installation method, which is not dependent on Linux versions or kernel versions and does not need to have a kernel patch applied.  Instructions for performing an RSUSB source code build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

YXGuan, 2022-12-19:
Will try

Thank you so much Marty!