realsenseai/librealsense #11107: using realsense camera for autonomous navigation using open cv

Pshcer, 2022-11-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (11)| 
                       | 

### Issue Description
I want to use the intel realsense camera to build an autonomous bot that detects the wall on both sides and moves through the middle of the wall.

are there any readily available scripts or what are the possibilities of accomplishing this task using OpenCV computer vision incorporating with dept data

MartyG-RealSense, 2022-11-15:
Hi @Pshcer  OpenCV is not typically used with RealSense for **wall following** applications that take account of left and right walls.  Navigation and obstacle avoidance with RealSense cameras is more commonly performed with ROS.    

My research of your question indicated that OpenCV wall following applications tended to use a technique called **Hough Transform**, like in the maze-solving paper at the link below.

https://campus-rover.gitbook.io/lab-notebook/gen4-reports/cvmaze

Leads to further resources can be found by googling for the term **opencv wall-following**

***************

In regard to a ROS based approach, an example of wall following with ROS and Python is here:

https://github.com/nimbekarnd/Wall-follower-in-ROS-using-Python

rohan3301, 2022-11-16:
ThankYou.

Pshcer, 2022-11-16:
thankyou

closed 2022-11-16 by Pshcer