realsenseai/librealsense #11112: D401 RealSense module disconnects as soon as firmware is uploaded in RealSense viewer

ChrisMute, 2022-11-15:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D4XX USB2 Recovery|
|**Product Line**|D400|
|**Firmware Update Id**|029323023871|

I open Intel RealSense Viewer v2.51.1 and connect new D401 module which shows as D4xx via USB2 on Windows 11. I download new firmware. Once the firmware download completes, instantly an error message shows Device disconnected, camera will not work? Thank you email:chris@foodphone.biz

MartyG-RealSense, 2022-11-15:
Hi @ChrisMute   Are you using the **V4** version of the Vision Processor D4 Board with your D401 camera module, please?  The V4 version is required because of its **Image Signal Processor (ISP) that enhances RGB data obtained from the depth sensor.  The D401 module cannot be used with Vision Processor D4 versions other than V4.

ChrisMute, 2022-11-15:
Hi Marty,
Thank you for your quick reply. Here is the version D4 board I recently got
from Digi Key:

- Is this the correct version?

 - If not can you give me Intel store or DigiKey part number to order , I
need quickly as we are prototyping in lab now 🙂 ?

- I never saw the D4 *V4 requirement * in D400 Datasheets , can you give me
a page or other documentation I can put in the project file?

Thank you, Chris Mutti

MartyG-RealSense, 2022-11-15:
Table 2-1 on page 19 of the current April 2022 edition of the data sheet document lists that the V4 version of D4 should be matched with the D401 module.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

The V4 board's 'Product Identifier Code' is M31706-100 and its 'Product Material Code' is 99AGAR.  The Digikry link for it is here:

https://www.digikey.de/en/products/detail/intel-realsense/82635DSASICBDV4/16265564

ChrisMute, 2022-11-15:
Hi Marty,

Thanks for this. I should have known better than to have checked the latest
version of the datasheet; I see it now!

One other question, on the D4 vision board, pictured here that I use for
the D430 module, can the interposer cable or board be connected to either
50 pin connector circled in this picture?

Thanks again for your help,

Chris Mutti Founder

MartyG-RealSense, 2022-11-16:
The interposer bridging the D4 to the depth module is attached to the upper of the two 50 pin connectors on the D4 board that is used with D430.  The lower connector was designed to have a tracking module attached to it.

MartyG-RealSense, 2022-11-21:
Hi @ChrisMute  Do you require further assistance with this case, please?  Thanks!

ChrisMute, 2022-11-21:
Hi MartyG,
I appreciate your support. Is the picture here the correct upper 50-pin
connector to the D430 module?

-Also, please provide the tracking module part number and link you are
referring to.

- is the tracking module connected to the lower 50-pin connector done with
the same interposer board as the D430 module?

Kind regards,

Christopher M. Mutti CEO Mngr.
       www.foodphone.biz
          ***@***.***
             (978) 857-6397
                *https://www.newenglandinnovation.org/innovation-stories
<https://www.newenglandinnovation.org/innovation-stories>*

MartyG-RealSense, 2022-11-22:
Images in comments that are sent by email do not appear in discussions.    Please come to the web page and insert the image into the comment writing box.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/11112#issuecomment-1322806311

ChrisMute, 2022-11-22:
[image][image]

MartyG-RealSense, 2022-11-22:
The data sheet document for the 400 Series cameras shows that when the V1 version of the D4 board is positioned horizontally, the depth module connector is the one at the top of the board and not the side one.

[image]

MartyG-RealSense, 2022-11-27:
Hi @ChrisMute  Do you require further assistance with this case, please?  Thanksl

ChrisMute, 2022-11-27:
Hi Martyg,

All set, thank-you for the excellent support.
.Chris Mutti

MartyG-RealSense, 2022-11-27:
You are very welcome.  :)

As you do not require further support, I will close this case.  Thanks again!

closed 2022-11-27 by MartyG-RealSense