realsenseai/librealsense #11115: Separating IMU pipeline in Unity

choe-hyun-ho, 2022-11-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Windows 11 | 
| Kernel Version (Linux Only)    |         | 
| Platform                                 | PC |
| SDK Version                            |  2.51.4348   | 
| Language                            |  Unity  2022.1.21f1    | 
| Segment			|                           | 

Hi Teams,
I reported IMU speed issue in #11111 , and get some successful result.
Because my final target is using D455 in Unity environment, I have tested in Unity Editor.

With single RSDevice, I successfully get video streams, and IMU streams (as you know, it is slow).

So, according to https://support.intelrealsense.com/hc/en-us/community/posts/360047049173-Community-contribution-for-using-multiple-cameras-in-RealSense-Unity-wrapper, I do followings.

- Copy the first RSDevice to second one
- For the first RSDevice, Remove Gyro & Accel elements in profile list
- For the second RSDevice, Only remains Gyro & Accel elements in profile list
- Enter the same device serial number for both RSDevice

But when I run the scean in Unity Editor, the second RSDevice never start properly.
Unity shows the following error.
```
ExternalException: rs2_pipeline_start_with_config(pipe:0000015D59593550, config:0000015D59593390)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <0f0f03ac46654106a5e240dad4689795>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <0f0f03ac46654106a5e240dad4689795>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)
```

How can I properly set up one physical device with two separate pipeline ?

MartyG-RealSense, 2022-11-16:
Hi @choe-hyun-ho  You could first test whether the problem is with two RsDevices trying to access the same camera by disabling the RsDevice that has the video stream so that only the remaining RsDevice with the IMU profile starts when the project is run.

The **Couldn't resolve requests** error usually occurs if a stream configuration has been requested that is not currently supported by the camera at the time that the project is run.  If you have tested the same IMU profile with one RsDevice and it works then this suggests that the configuration is correct if you are using the same frequencies for Accel and Gyro on the second RsDevice.

I would recommend double-checking the frequencies on the second RsDevice.  For example, on older D455 units the default frequencies for accel and gyro may be 63 / 200, but on a newer D455 unit the defaults may be 100 / 200 - as described at https://github.com/IntelRealSense/librealsense/issues/11089#issuecomment-1311940940

Also check that the formats on the second RsDevice are set to **Motion Xyz 32f** for accel and gyro like in the image below.

[image]

choe-hyun-ho, 2022-11-16:
Thanks, MartyG.

The problem is caused by incorrect framerate for each IMU unit.
I changed values to Accel=100, Gyro=200, and now it works fine.

BTW, even I separated pipeline, I feel no difference to the single pipeline case.
There are lots of the same timestamp frame for both Accelo & Gyro.

I am using the following IMU script for gathering data. Any advice?
```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using Intel.RealSense;
using Intel.RealSense.Math;
using System;
using System.Threading;
using UnityEngine.Events;
using System.Runtime.InteropServices;

public class IMU : MonoBehaviour
{
    private static TextureFormat Convert(Format lrsFormat)
    {
        switch (lrsFormat)
        {
            case Format.Z16: return TextureFormat.R16;
            case Format.Disparity16: return TextureFormat.R16;
            case Format.Rgb8: return TextureFormat.RGB24;
            case Format.Rgba8: return TextureFormat.RGBA32;
            case Format.Bgra8: return TextureFormat.BGRA32;
            case Format.Y8: return TextureFormat.Alpha8;
            case Format.Y16: return TextureFormat.R16;
            case Format.Raw16: return TextureFormat.R16;
            case Format.Raw8: return TextureFormat.Alpha8;
            case Format.Disparity32: return TextureFormat.RFloat;
            case Format.Yuyv:
            case Format.Bgr8:
            case Format.Raw10:
            case Format.Xyz32f:
            case Format.Uyvy:
            case Format.MotionRaw:
            case Format.MotionXyz32f:
            case Format.GpioRaw:
            case Format.Any:
            default:
                throw new ArgumentException(string.Format("librealsense format: {0}, is not supported by Unity", lrsFormat));
        }
    }

    public RsFrameProvider Source;

    public Stream _stream;
    public Format _format;
    public int _streamIndex;

    Predicate<Frame> matcher;

    public Vector ImuData;
    public double Timestamp;

    private void Start()
    {
        Source.OnStart += OnStartStreaming;
        Source.OnStop += OnStopStreaming;
    }

    protected void OnStopStreaming()
    {
        Source.OnNewSample -= OnNewSample;
    }
    void OnNewSample(Frame frame)
    {
        try
        {
            if (frame.IsComposite)
            {
                using (var fs = frame.As<FrameSet>())
                using (var f = fs.FirstOrDefault<MotionFrame>(matcher))
                {
                    if (f != null)
                    {
                        ImuData = f.MotionData;
                        Timestamp = f.Timestamp;
//Debug.Log("X=" + ImuData.x + " Y=" + ImuData.y + " Z=" + ImuData.z);
                        f.DisposeWith(fs);
                    }
                    return;
                }
            }
            if (!matcher(frame))
                return;
            ImuData = ((MotionFrame)frame).MotionData;
            Timestamp = frame.Timestamp;
//Debug.Log("X=" + ImuData.x + " Y=" + ImuData.y + " Z=" + ImuData.z);
        }
        catch (Exception e)
        {
            Debug.LogException(e);
            // throw;
        }
    }
    public void OnStartStreaming(PipelineProfile activeProfile)
    {
        matcher = new Predicate<Frame>(Matches);
        Source.OnNewSample += OnNewSample;
    }
    private bool Matches(Frame f)
    {
        using (var p = f.Profile)
            return p.Stream == _stream && p.Format == _format && p.Index == _streamIndex;
    }
}
```

MartyG-RealSense, 2022-11-16:
In the past some users of the RealSense Unity wrapper have experienced long delays when stopping IMU streaming.  This is discussed in https://github.com/IntelRealSense/librealsense/issues/5993 and https://github.com/IntelRealSense/librealsense/issues/9278

choe-hyun-ho, 2022-11-17:
I re-tested codes with measuring frame rate.
Before separating IMU pipeline, I got 30 fps for each IMU, and after separating, I got 190~200 fps per each.

Thanks for the support, MartyG.

MartyG-RealSense, 2022-11-17:
That's great to hear that you were successful, @choe-hyun-ho - thanks very much for the update!