realsenseai/librealsense #11116: Face recognition by IR image (OpenCV & RealSense D435 & Visual Studio 2017)

Vermilion016, 2022-11-17:
I am learning ReaSense D435 and Visual Studio (C++).
I now have two codes.
A code that reflects the RealSense infrared stream (IR image) and a code that can recognize faces with a PC camera and OpenCV.
We are researching whether facial recognition using IR images can be achieved by combining these two.
If there is a better way, please let me know.
Below is the code and references.

`#include "opencv2/opencv.hpp"
#include "opencv2/highgui.hpp"
#include <vector>

using namespace cv;
using namespace std;

int main(void) {

	//Device Open
	VideoCapture cap(0);

	//Check if the camera device was opened normally.
	if (!cap.isOpened())
	{
		return -1;
	}

	//Frame taken
	Mat frame;

	//Cascade classifier storage location
	CascadeClassifier cascade;
	cascade.load("C:/opencv/sources/data/haarcascades/haarcascade_frontalface_alt.xml");

	//Where to store contour information
	vector<Rect> faces;

	//loop
	while (1)
	{
		//
		cap >> frame;

		cascade.detectMultiScale(frame, faces, 1.1, 3, 0, Size(20, 20));

		for (int i = 0; i < faces.size(); i++)
		{
			//Enclose the detected face with a red rectangle
			rectangle(frame, Point(faces[i].x, faces[i].y), Point(faces[i].x + faces[i].width, faces[i].y + faces[i].height), Scalar(0, 0, 255), 3);
		}

		//image
		imshow("win", frame);
	
		const int key = waitKey(1);

		if (key == 'q'/*113*/)
		{
			break;//whileループから抜ける．
		}

	}
	//Window close
	destroyAllWindows();

	return 0;
}`
`#include <librealsense2/rs.hpp>
#include <opencv2/opencv.hpp>
#include <iostream>
#include <stdio.h>
// https://github.com/IntelRealSense/librealsense/issues/1140

int main(int argc, char * argv[])
{
	int width = 1280;
	int height = 720;
	int fps = 30;
	rs2::config config;
	config.enable_stream(RS2_STREAM_INFRARED, width, height, RS2_FORMAT_Y8, fps);

	//パイプライン開始
	rs2::pipeline pipeline;
	rs2::pipeline_profile pipeline_profile = pipeline.start(config);

	while (1) // アプリケーションが生きている
	{
		// フレームセット取得
		rs2::frameset frameset = pipeline.wait_for_frames();

		//赤外線フレームを取得
		rs2::video_frame ir_frame = frameset.get_infrared_frame();

		cv::Mat dMat_left = cv::Mat(cv::Size(width, height), CV_8UC1, (void*)ir_frame.get_data());

		cv::imshow("img", dMat_left);
		char c = cv::waitKey(1);
		if (c == 's')
		{

		}
		else if (c == 'q')
			break;
	}

	return EXIT_SUCCESS;
}`
[https://github.com/IntelRealSense/librealsense/issues/1140](url)
[https://shiyotsuki.hatenablog.com/entry/virtual_youtuber_5](url)

MartyG-RealSense, 2022-11-17:
Hi @Vermilion016  A more common approach to face recognition would be to use the RGB camera for face analysis, and to also project **infrared light** and perhaps additionally a pattern of dots in order to illuminate the face in low-light conditions and aid analysis.  This approach is taken by the RealSense F450 / F455 facial authentication device (which is now End of Life / retired).

https://www.intelrealsense.com/facial-authentication/

[image]

[image]

You could make use of the D435 camera's infrared projector (which is a separate component from the infrared sensors) without having to enable an infrared stream.   You could enable it in your script with the SDK instruction **RS2_OPTION_EMITTER_ENABLED**.  Below is an example of C++ code for doing so.

```
rs2::pipeline pipe; 
rs2::pipeline_profile selection = pipe.start();
rs2::device selected_device = selection.get_device();
auto depth_sensor = selected_device.first<rs2::depth_sensor>();

if (depth_sensor.supports(RS2_OPTION_EMITTER_ENABLED))
{
depth_sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 1.f);
}
```

Vermilion016, 2022-11-17:
I'm sorry.
It was resolved around 3pm.
Thank you for your cooperation.

closed 2022-11-17 by Vermilion016

MartyG-RealSense, 2022-11-17:
It's no trouble at all.  Thanks very much for the update!