realsenseai/librealsense #11117: rs-enumerate-devices command on linux spits out "no devices detected. Is it plugged IN?"

VishnuRamTK, 2022-11-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux built from petalinux 2021.1 | 
| Kernel Version (Linux Only)    |  5.10.0-xilinx-v2021.1                      | 
| Platform                                 | Kria KV260 FPGA board  |
| SDK Version                            |  { latest version not sure which }                          | 
| Language                            |  {Cpp/python }                          | 
| Segment			|  {Student project }                          | 

### Issue Description
I was trying to interface the Intel realsense 435i Camera with Kria KV260 FPGA board. I built the Linux system with the help of petalinux version 2021.1 tool and included the UVC and other self-hosted package libraries for support. After the build i downloaded the lirealsense repository from Github following which i updated the udev rules and ran the cmake command in a separate build folder. After everything was built i exported the LD_Library_Path to access the compiled real sense libraries(export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/usr/local/lib64/"). when i then ran the rs-enumerate-devices command i get a message in the screen that the device is not detected and asking for the prompt if the device is pluggedin. Any directions in solving the issue will be greatly helpful as i don't have much time to submit my project.

MartyG-RealSense, 2022-11-17:
Hi @VishnuRamTK  There is a guide at the link below to installing the Realsense SDK on Kria KV260 FPGA board.and Petalinux version 2021.1.  The web page has some formatting problems, so scroll down through the empty areas of the page to see the text of the guide.

https://www.hackster.io/jack-bonnell2/realsense-support-for-kria-kv260-petalinux-2021-1-f5dbe3

For FPGA devices running Petalinux though, I would recommend building the SDK from source code with CMake using the **RSUSB backend** installation method, or its predecessor **LibUVC backend**.  These two methods bypass the Linux kernel and so are not dependent on Linux versions or kernel versions and do not require kernel patching.

Installtion with the LibUVC backend method on a Petalinux device is discussed at https://github.com/IntelRealSense/librealsense/issues/3040

Instructions for the more modern RSUSB backend approach can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

VishnuRamTK, 2022-11-18:
Thank you very much for your quick reply @MartyG-RealSense. I am sorry that i didnt mention the references on how i build my linux system from petalinux. I have followed the exact same tutorial from Jack Bonnell as above with minor variations after facing series of errors. This time i tried installing the Librealsense SDK 2.51.1 from source.tar.gz file as directed in your comment #9931(comment). The **cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true** command could succesfully extract build files in to the build directory but once i run **sudo make uninstall && make clean && make -j8 && sudo make install** i get stuck up with the  following process -

[ 17%] Building CXX object CMakeFiles/realsense2.dir/src/ds5/advanced_mode/rs_advanced_mode.cpp.o

Tested: int send(SOCKET, char *, unsigned int, int)
-- Performing Test curl_cv_func_send_test
-- Performing Test curl_cv_func_send_test - Failed

I am yet to try with LibUVC. Will try it in meantime.

MartyG-RealSense, 2022-11-18:
Using a **-j** parameter to accelerate the build process can cause random build crashes on some devices such as single-board computers.  The build may be more stable if -j8 is left out of the make instruction, though could take significantly longer to complete.

`sudo make uninstall && make clean && make && sudo make install`

Please do let me know how LibUVC works for you too.  Good luck!

VishnuRamTK, 2022-11-18:
Hi @MartyG-RealSense  When i ran with  **cmake ../ -DFORCE_LIBUVC=true. -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**  flag and consectively **sudo make uninstall && make clean && make -j8 && sudo make install** i get stuck up at another point -

HEAD is now at 2f33be817 RELEASE-NOTES: synced
[ 17%] No patch step for 'libcurl'
[ 17%] No update step for 'libcurl'
[ 17%] Performing configure step for 'libcurl'

Performing Test OPT_Wcast_align - Success
-- Performing Test OPT_Wtype_limits
-- Performing Test OPT_Wtype_limits - Success
-- Performing Test OPT_Wold_style_declaration

I ll now try your suggestion and update you soon.

VishnuRamTK, 2022-11-20:
Thanks a lot @MartyG-RealSense. I used **sudo make uninstall && make clean && make && sudo make install** to install and the SDK could be succesfully installed. When the camera was connected and rs-enumerate-devices command was run i could see the camera getting detected.

closed 2022-11-20 by VishnuRamTK

MartyG-RealSense, 2022-11-20:
That's great to hear, @VishnuRamTK - thanks very much for the update!

VishnuRamTK, 2022-11-22:
Hi @MartyG-RealSense i could not see the pyrealsense.so file anywhere after the installation. The readme.md document in the page https://github.com/IntelRealSense/librealsense/tree/251/wrappers/python suggests alternative ways to install pyrealsense however nothing works for me. Apparently the bdist_wheel argument for setup.py is obsolete or not found. Should i once again reinstall the the whole SDK as suggested in **building from source** section on this page?

MartyG-RealSense, 2022-11-23:
Using setup.py to install the Python wrapped with a wheel file may not work for the Kria KV260 FPGA board as the whl files are designed for **amd64** computer architectures such as desktop and laptop PCs.

The wrapper can alternatively be installed by building librealsense from source code and including CMake build flags to install pyrealsense2 at the same time as the SDK, or by building the wrapper only from source ,(as you mentioned) after already having installed librealsense.

VishnuRamTK, 2022-11-24:
Hi @MartyG-RealSense. I built the sdk from source with relevant changes mentioned in the building from source section and things worked out. I got the pyrealsense2.so file but with a different name but renamed it and copied it along with librealsense.so to the location of the py_viewer_depth.py script following which i ran it. I could see the color and depth streams from the camera. Thanks a lot for the support.

MartyG-RealSense, 2022-11-24:
You are very welcome, @VishnuRamTK - it's great to hear that you succeeded.  Thanks very much for the update!