realsenseai/librealsense #11118: Align.Process to upscaling IR and Depth to Color.

rajrajeswaran, 2022-11-17:
Camera Model: Intel D415
Firmware: 05.12.07.100
Operating System: Windows 10
Platform: PC
SDK version: C# RealSense SDK 2.38
Language: C#
Segment: Windows Forms application.

I am currently using Intel RealSense C# SDK version 2.38 with Intel D415 sensor.  Trying to understand why the following code does not upscale the IR and Depth frame to 1080 P.  Any suggestion will be helpful.
```
using (frames)
                            {
                                foreach (var frame in frames)
                                {
                                    using (frame)
                                    {
                                        using (var p = frame.Profile)
                                        {
                                            Logger.Info(
                                                $"    {p.Stream} {p.Format,4} #{frame.Number} {frame.TimestampDomain} {frame.Timestamp:F2}");
                                            if (p.Stream == Stream.Color)
                                            {
                                                OnColorFrame(frame);
                                                LastFrameTime = DateTime.UtcNow;
                                            }
                            
                                            if (p.Stream == Stream.Infrared)
                                            {
                                                using (var aligned = new Align(Stream.Color).Process(frame))
                                                {
                                                    OnInfraRedFrame(aligned);
                                                }
                                            }
                            
                                            if (p.Stream == Stream.Depth)
                                            {
                                                using (var aligned = new Align(Stream.Color).Process(frame))
                                                {
                                                    OnDepthFrame(aligned);
                                                }
                                            }
                                        }
                                    }
                                }
                            }

```

rajrajeswaran, 2022-11-21:
> Hi @rajrajeswaran Infrared cannot be aligned to Color in this way with an Align(...).Process instruction, as one of the two stream types in this kind of alignment has to be Depth (such as Depth to Color, or Color to Depth).
> 
> [#1556 (comment)](https://github.com/IntelRealSense/librealsense/issues/1556#issuecomment-389061477) provides advice regarding an alternative method for how infrared and color might be aligned. There is not an example available of C# code for doing so though.
> 
> #5093 is a case that explores aligning all three streams as the above link suggests. Again, there is not a C# example for doing so, unfortunately.

@MartyG-RealSense  as always, Thank you for the quick response.  I will review those posts.  You may close this request.  Thanks again.

rajrajeswaran, 2022-11-21:
@MartyG-RealSense as always, Thank you for the quick response. I will review those posts. You may close this request. Thanks again.

closed 2022-11-21 by rajrajeswaran

MartyG-RealSense, 2022-11-22:
You are very welcome, @rajrajeswaran -thanks very much for the update!