realsenseai/librealsense #11121: About distance of RealSense D435

Vermilion016, 2022-11-19:
I want to add a program to display the distance between the center of the screen and the camera in the code below with realsense D435.
Is there a better way?

`#include "opencv2/opencv.hpp"
#include "opencv2/highgui.hpp"
#include <vector>
#include <librealsense2/rs.hpp>
#include <iostream>
#include <stdio.h>
using namespace cv;
using namespace std;
int main(int argc, char * argv[])
{
	int width = 640;
	int height = 360;
	int fps = 15;
	float dist_to_center;
	rs2::config config;
	config.enable_stream(RS2_STREAM_INFRARED, width, height, RS2_FORMAT_Y8, fps);
	rs2::pipeline pipeline;
	rs2::pipeline_profile pipeline_profile = pipeline.start(config);
	CascadeClassifier cascade;
	cascade.load("C:/opencv/sources/data/haarcascades/haarcascade_frontalface_alt.xml");
	vector<Rect> faces;
	while (1)
	{
		rs2::frameset frameset = pipeline.wait_for_frames();
		rs2::video_frame ir_frame = frameset.get_infrared_frame();
		cv::Mat dMat_left = cv::Mat(cv::Size(width, height), CV_8UC1, (void*)ir_frame.get_data());
		cascade.detectMultiScale(dMat_left, faces, 1.1, 3, 0, Size(20, 20));
		rs2::depth_frame depth = frameset.get_depth_frame();
		for (int i = 0; i < faces.size(); i++)
		{
			rectangle(dMat_left, Point(faces[i].x, faces[i].y), Point(faces[i].x + faces[i].width, faces[i].y + faces[i].height), Scalar(0, 0, 0), 3);
		}
		imshow("img", dMat_left);
		char c = cv::waitKey(1);
	}
	return EXIT_SUCCESS;
}`

MartyG-RealSense, 2022-11-19:
Hi @Vermilion016  A C++ / OpenCV script at https://github.com/IntelRealSense/librealsense/issues/6239#issuecomment-615249398 for getting the pixel at the center of an object may be a helpful reference.

Vermilion016, 2022-11-19:
I'm sorry. I failed.
I actually tried the sample code in the link below. It worked it, but I couldn't union it with the original code.
'p.start();'
'pipeline.start();'
It seems that it was useless this part.

[https://dev.intelrealsense.com/docs/rs-hello-realsense](url)

MartyG-RealSense, 2022-11-19:
There is no need to be sorry.  :)

If you attempted to merge the rs-hello-realsense project into your own project then it may have failed to build because of the CMakeLists.txt file of rs-hello-realsense not having the necessary details to locate the librealsense library files.  The link below has a version of rs-hello-realsense with a more complex CMakeLists.txt file that can work as a standalone build.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

Vermilion016, 2022-11-19:
I'see. I will try.
Thank you very much.

MartyG-RealSense, 2022-11-27:
Hi @Vermilion016  Do you require further assistance with this case, please?  Thanksl

Vermilion016, 2022-11-27:
This method was unsuccessful, but other methods may be substituted.
Thank you for your cooperation. Mr. Marty G.

MartyG-RealSense, 2022-11-27:
Do you need further suggestions about how to solve your problem, please?

MartyG-RealSense, 2022-12-04:
Case closed due to no further comments received

closed 2022-12-04 by MartyG-RealSense