realsenseai/librealsense #11122: A question about "Couldn't resolve requests"

ragidcat, 2022-11-19:
i search for a lot ,but a question is still not resolved.
my camera is realsense L515,but when i run the code like this 
(pipeline = rs.pipeline() 
config = rs.config()
.
.
.
pipe_profile = pipeline.start(config)
),there will be a problem is "Couldn't resolve requests"

And i use the usb3.0,i do not konw how to resolve it ,help me

MartyG-RealSense, 2022-11-19:
Hi @ragidcat  The error **Couldn't resolve requests** means that the stream configuration defined in the config lines of the script could not be provided, either because of a typing mistake in the code or because that particular stream configuration is not supported by the camera.

An example of an unsupported camera mode would be if the config lines tried to define an FPS speed other than 30 for depth or infrared, as the L515 only supports 30 FPS for depth / infrared.  Another example would be setting 848x480 resolution, as L515 does not support that resolution.

Could you paste the 'config' lines of your script in a comment, please?  Thanks!

MartyG-RealSense, 2022-11-27:
Hi @ragidcat  Do you require further assistance with this case, please?  Thanksl

closed 2022-12-04 by MartyG-RealSense