realsenseai/librealsense #11123: realsense

Keanu-Re, 2022-11-20:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|134322075122|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|134523062803|
|**Firmware Update Id**|134523062803|

Please provide a description of the problem

Keanu-Re, 2022-11-20:
[image]

[image]

After these two calibrations, the depth measurement is not accurate and cannot be calibrated again。
[image]

It is useless to try to reset the hardware and re brush the firmware
[image]

MartyG-RealSense, 2022-11-20:
Hi @Keanu-Re  That particular **Calibration Failed** message may occur if the lighting level is low in the area that the camera is located in.  Increasing the illumination of the wall in that area or moving the camera to a more brightly lit location may help.  

Alternatively, you may be able to instead improve the calibration by resetting the camera in the RealSense Viewer to its factory-new default calibration using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

closed 2022-11-20 by Keanu-Re