realsenseai/librealsense #11124: sparse point cloud noise in backside of camera direction L515

robopassio, 2022-11-20:
Hello,

I'm using L515 to capture point cloud (PCL) data at resolution 1024 x 768, 30 fps.
The captured PCL looks great at regions visible to the camera eye.
For the backside (invisible to the camera) regions, I expect no PCL will be obtained. However, sparse noise appears in those regions. The density of those patterns is arbitrary. So it's impossible to filter out using some filters.

[image]

Is there any way to address this issue?
Thanks.

MartyG-RealSense, 2022-11-20:
Hi @robopassio  A way to remove sparse data in PCL without having to filter by depth values would be to apply a **StatisticalOutlierRemoval** filter to exclude loose points that are not joined to a large area.  A script example of using this PCL filter with the L515 camera model is at https://github.com/IntelRealSense/librealsense/issues/9658#issuecomment-914647767

        ```
//Use a filter to reduce noise
	pcl::StatisticalOutlierRemoval<pcl::PointXYZ> statFilter;
	statFilter.setInputCloud(cloud_downsampled);
	statFilter.setMeanK(10);
	statFilter.setStddevMulThresh(0.2);
	statFilter.filter(*cloud);
```

robopassio, 2022-11-20:
As I mentioned, applying an outlier removal will not be a good approach due to the variation of pcl density. 
So I'm looking for a camera configuration (before reading data) that can be addressed that. Is that possible?
Also, does that kind of noise always happen in the practice of LiDAR sensors?

robopassio, 2022-11-20:
[This](https://github.com/IntelRealSense/librealsense/issues/7117#issuecomment-694684795) issue is quite related to mine. However, the method to resolve that is not clear. Could you please give me few suggestions?

MartyG-RealSense, 2022-11-20:
I see, you mean the density of the points is changing, meaning that they will not consistently be outliers.  Thanks very much for the clarification.

Whilst I cannot speak for the performance of non-RealSense lidar devices, there are certain scenarios where the quality of results from an L515 lidar depth camera may reduce.  This could include natural sunlight entering through a window (as the L515 is sensitive to IR light sources) or pointing the camera directly at a smooth surface (tilting the camera at an angle can provide better results for such surfaces).  The L515 will also tend to look through transparent objects such as empty drinks bottles and pick up the solid  depth detail behind the bottle.

An L515 user guide document with image improvement tips can be obtained from the link below.

Regarding https://github.com/IntelRealSense/librealsense/issues/7117#issuecomment-694684795 I could not determine from the comments which filtering techniques the RealSense user was applying.

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

Setting the L515's Visual Preset camera configuration as **Short Range** may help to reduce noise.  C++ code for doing so is below.

```
auto sensor = profile.get_device().firstrs2::depth_sensor();
sensor.set_option(rs2_option::RS2_OPTION_VISUAL_PRESET,
rs2_l500_visual_preset::RS2_L500_VISUAL_PRESET_SHORT_RANGE);
```

robopassio, 2022-11-20:
> Regarding https://github.com/IntelRealSense/librealsense/issues/7117#issuecomment-694684795 I could not determine from the comments which filtering techniques the RealSense user was applying.

I think this issue is quite popular in recording PCL using L515. I think it was resolved by a postprocessing step. Hope you or other guys (maybe @RealSenseSupport ) could suggest a specific technique to resolve that.

MartyG-RealSense, 2022-11-20:
The only depth post-processing filter supported by the RealSense SDK for L515 is the **Temporal** filter.

[image]

If the Temporal filter has its **Filter Smooth Delta** setting increased to its maximum of '100' then it may result in a reduction of black holes in the far background and a little less fluctuation of values.

**Before**

[image]

**After**

[image]

MartyG-RealSense, 2022-11-20:
Enabling an option called **Invalidation Bypass** can also fill in holes.

**Invalidation Bypass disabled**

[image]

**Invalidation Bypass enabled**

[image]

MartyG-RealSense, 2022-11-20:
Could you provide further information about the following statement from your opening comment, please: "The captured PCL looks great at regions visible to the camera eye.  For the backside (invisible to the camera) regions, I expect no PCL will be obtained". 

Are the 'backside'and 'invisible' regions areas of the scene that are not being picked up by the depth sensing for some reason?  It looks as though there should be a solid bathroom there but for some reason a lot of the detail is not being picked up.

It would be very helpful if you could provide an RGB image of the scene so that I can see whether there are lighting conditions or materials that might be making it difficult for those areas of the scene to have depth detail detected.  

Thanks very much for your patience!

MartyG-RealSense, 2022-11-27:
Hi @robopassio  Do you require further assistance with this case, please?  Thanksl

robopassio, 2022-11-29:
Hi @MartyG-RealSense, 
> Are the 'backside'and 'invisible' regions areas of the scene that are not being picked up by the depth sensing for some reason? 

Yes, the backside of the object, that is not visible to the camera, should not have any PCL. However, as you can see from the posted image, the sparse noise happen. Also, refer to the [comment](https://github.com/IntelRealSense/librealsense/issues/7117#issuecomment-694684795), it is called as "tails" of the object. My issue is very similar to that one. We don't want to have any noise in the edge or "tail" PCL at invisible regions to the camera. Is that possible?

MartyG-RealSense, 2022-11-29:
Perhaps you could reduce background noise by increasing the value of the L515's **RS2_OPTION_CONFIDENCE_THRESHOLD** setting from its default value of '1'.  This setting is the confidence level threshold used by the depth algorithm pipe to set whether a pixel will get a valid range or will be marked with invalid range.

Here is an example use:

`depth_sensor.set_option(RS2_OPTION_CONFIDENCE_THRESHOLD, 0);`

**************

You can see the effect on distant noise in the RealSense Viewer by adjusting the Confidence Threshold slider in the 'L500 Depth Sensor > Controls' section of the Viewer's options side-panel.

[image]

closed 2022-12-04 by robopassio